calyxir/calyx #1227: Support Inline Verilog for Primitive Definitions

andrewb1999, 2022-11-03:
It would be nice to support inline verilog for primitive definitions instead of requiring a separate verilog file. For core primitives it makes sense to have the verilog separate for many reasons, but when generating memory primitives from Calyx having inline verilog would simplify the process a lot.

Specifically, primitives in CIRCT are currently required to be HWModuleExtern operators that function very similarly to the current Calyx primitives, the definitions are assumed to be in another file. It would be preferable for primitives in CIRCT to be HWModuleLike interface operators so we can just generate the primitives in CIRCT using the HW dialect if desired. Exporting this inline in a single Calyx file would be much easier than writing out a separate verilog file for the primitives and ensuring they are in the correct directories for simulation.

calebmkim, 2023-01-19:
Just to double check I'm understanding this correctly, I was thinking that we want to support the Calyx file to look something like this: 
```
verilog {
    module verilog_module #( ...) 
    (define your own verilog modules here)
}
extern {
      ...
}
(rest of the calyx file here)
```
Sorry if this is a basic question, just making sure I'm understanding correctly.

sampsyo, 2023-01-19:
Something like this seems like it would work, although the prescription here is very open ended! FWIW, another thing I could imagine working is nesting the Verilog code inside the Calyx primitive definition. Something like:

```
primitive seq_mem_d1[WIDTH, SIZE, IDX_SIZE](
  @clk clk: 1,
  ...
) -> (
  @stable out: WIDTH,
  ...
) {
  // Verilog "guts" go here.
}
```

That admittedly seems much more annoying to handle, because you'd have to generate the Verilog module declaration according to the Calyx signature. But it would have the semi-nice upside of reducing some duplication.

rachitnigam, 2023-01-19:
Yup, @sampsyo has it right!