calyxir/calyx #1228: Interpreter Output doesn't match verilator and icarus verilog output

susan-garry, 2022-11-03:
The issue seems to have something to do with using the `invoke` keyword when passing memories by reference "the hard way." I initially noticed the issue when running `exine depth -a -r test/k.og` from inside the root directory of the [pollen](https://github.com/cucapra/pollen) repository, but this is the simplest example where I notice a difference in output:

[k3.futil.txt](https://github.com/cucapra/calyx/files/9933443/k3.futil.txt)
[k3.data.txt](https://github.com/cucapra/calyx/files/9933522/k3.data.txt)
## Verilog and Icarus Verilog
Command: `fud e --to dat --through verilog -s verilog.data k3.data.txt k3.futil.txt`
Output:
```
{
  "cycles": 8,
  "memories": {
    "depth_output": [
      1
    ],
    "path_ids1": [
      1
    ]
  }
}
```
## Interpreter
Command: `fud e --to interpreter-out -s verilog.data k3.data.txt k3.futil.txt`
Output: 
```
{
  "main": {
    "depth_output": [
      2
    ],
    "path_ids1": [
      1
    ]
  }
}
```

The issue isn't simply with passing registers by reference, since the interpreter produces the expected output for this [simple example](https://github.com/cucapra/calyx/files/9933562/simple-invoke.futil.txt).

susan-garry, 2022-11-03:
I think that the interpreter may be calling each invoke statement twice. This would also make sense of the above output, the output of running `[calyx/examples/futil/pass-in-register.futil](https://github.com/cucapra/calyx/blob/master/examples/futil/pass-in-register.futil)` through the interpreter.

EclecticGriffin, 2022-11-04:
I'll take a look later today and see what I can suss out

EclecticGriffin, 2022-11-04:
Okay I know what the problem is, unfortunately it is kinda complicated so this might take me a little while to properly work out

PR #1237 (2022-11-08): [Interpreter] turn off group assignments the same cycle done becomes high
This closes #1228. @susan-garry do let me know if other bugs emerge, this fixes the particular error with the files you identified which hopefully was the only issue.

Sundry details:
- After clippy bullying me the InterpreterResult alias now uses a boxed `InterpreterError` since the error type is 200 bytes (!!). Apparently Calyx `Id`s are 72 bytes a piece.
- Some minor changes to the root toml, mainly switching to thin `lto` and making the interpreter use incremental builds for the release profile
- minor clippy lints
- [... cut]
closed 2022-11-08 by EclecticGriffin