calyxir/calyx #1239: Assignments incorrectly removed in comb-prop

andrewb1999, 2022-11-08:
Just found a bug in comb-prop where some continuous assignments are completely removed when they should not be. Here is a simple example:
```
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1, in: 32) -> (@done done: 1, out: 32) {
  cells {
    wire1 = std_wire(32);
  }
  wires {
    wire1.in = in;
    out = wire1.out;
    done = 1'b1;
  }
  control {
  }
}
```
After comb-prop the result is:
```
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1, in: 32) -> (@done done: 1, out: 32) {
  cells {
    wire1 = std_wire(32);
  }
  wires {
    done = 1'd1;
  }

  control {}
}
```
I would expect a correct result to include a continuous assignment `out = in` to be retained.

rachitnigam, 2022-11-08:
Super weird! If you pass `-x comb-prop:no-eliminate`,  the pass should mark assignments it thinks are dead with `@dead` instead of removing them: https://github.com/cucapra/calyx/blob/master/calyx/src/passes/comb_prop.rs#L139

andrewb1999, 2022-11-08:
> Super weird! If you pass `-x comb-prop:no-eliminate`, the pass should mark assignments it thinks are dead with `@dead` instead of removing them: https://github.com/cucapra/calyx/blob/master/calyx/src/passes/comb_prop.rs#L139

Yeah I figured this out, it outputs the following:
```
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1, in: 32) -> (@done done: 1, out: 32) {
  cells {
    wire1 = std_wire(32);
  }
  wires {
    @dead out = in;
    @dead out = in;
    done = 1'd1;
  }

  control {}
}
```
For some reason both of the assignments from in to out are marked dead.

sampsyo, 2022-11-12:
Very interesting indeed! I was curious and checked what happened if we use a constant instead of a wire:

```
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1, in: 32) -> (@done done: 1, out: 32) {
  cells {
    const1 = std_const(32, 1337);
  }
  wires {
    out = const1.out;
    done = 1'b1;
  }
  control {
  }
}
```

…and the `out =` assignment is _not_ dead here. So it seems like this may be specific to the way the pass handles `std_wire`.

commit 1d64f9e1 (2022-11-15): Fix #1239 (#1267)
closed 2022-11-15 by rachitnigam