calyxir/calyx #1241: Icarus verilog differing from verilator

susan-garry, 2022-11-08:
For the following files, [`k.futil`](https://github.com/cucapra/calyx/files/9964353/k.futil.txt) and [`k.data`](https://github.com/cucapra/calyx/files/9964352/k.data.txt), `fud e --to dat --through verilog -s verilog.data k.data k.futil` works as expected, while `fud e --to --dat --through icarus-verilog -s verilog.data k.data k.futil` produces the following error:

```
[fud] ERROR: Invalid Numeric Type: Memory contains the value: `0xx', which is uninitialized.
This happens when the Calyx design attempts to read a port that is not connected to anything.
Try to generate a VCD file using `--to vcd` and look for signals that are not driven in a given cycle.
```

Is this an issue with icarus verilog itself, or with the input we are generating for icarus verilog?

rachitnigam, 2022-11-08:
I provided an explanation here: https://github.com/cucapra/pollen/discussions/15#discussioncomment-4090082

The problem is some port is likely undriven and neither Verilator nor interpreter would provide an error when that happens. However, it points to a real bug in the program.

rachitnigam, 2022-11-08:
With the changes in this PR: https://github.com/cucapra/calyx/pull/1242, I get the following output from icarus:
```
% fud e --to dat --through icarus-verilog -s verilog.data k.json k.futil -s futil.flags ' -p no-opt' -q
[fud] ERROR: Memory contains the value: `0xx', which is uninitialized. This happens when the Calyx design attempts to read a port that is not connected to anything. Try to generate a VCD file using `--to vcd` and look for signals that are not driven in a given cycle.
{
  "cycles": 52,
  "memories": {
    "depth_output": [
      2,
      1
    ],
    "path_ids1": [
      1,
      2
    ],
    "path_ids2": [
      2,
      0
    ],
    "paths_to_consider1": [
      0,
      1,
      1
    ],
    "paths_to_consider2": [
      0,
      1,
      1
    ],
    "uniq_output": [
      "2",
      "x"
    ]
  }
}
```

Note how `uniq_output` contains an "x" value. This is the problem that `fud` was complaining about. Somewhere along the computation that generates `uniq_output[1]`, there is an undriven port (i.e., a port without an assignment) causing the problem.

rachitnigam, 2022-11-08:
@susan-garry can you close the issue once you validate that the bug is caused from generated code?

susan-garry, 2022-11-09:
Yep, this was an issue with me not instantiating all of the memories. Thank you for all your help!

closed 2022-11-09 by susan-garry