calyxir/calyx #1249: `compile-sync` does not handle group enables with surrounding `seq` block

rachitnigam, 2022-11-12:
When I change `tests/correctness/sync/sync-dot-product.futil` to not have `seq` around the `no_op` blocks:
```patch
diff --git a/tests/correctness/sync/sync-dot-product.futil b/tests/correctness/sync/sync-dot-product.futil
index db61f72f..01a828e5 100644
--- a/tests/correctness/sync/sync-dot-product.futil
+++ b/tests/correctness/sync/sync-dot-product.futil
@@ -224,11 +224,8 @@ component main () -> () {
               }
               @sync(2) val_to_reg_1;
             }
-          }
-          else {
-              seq {
-                @sync(1) no_op;
-              }
+          } else {
+            @sync(1) no_op;
           }
         }
       }
```

And run:
```
cargo run -- tests/correctness/sync/sync-dot-product.futil -p validate -p compile-sync -m file
```

The generated program leaves `@sync no_op` unchanged which is incorrect

paili0628, 2022-11-12:
I see. Right now the compile-sync pass does not parse enable controls directly under the par block. They have to be wrapped inside a seq to be compiled into barriers.

paili0628, 2022-11-12:
I will change it so that it adds barriers properly to "direct" enables too

rachitnigam, 2022-11-12:
Makes sense! Thanks!

rachitnigam, 2022-12-10:
@paili0628 was this fixed by your recent changes?

paili0628, 2022-12-10:
yes I suppose so, is anything wrong?

rachitnigam, 2022-12-10:
Nope, we should just close the issue of the problem is fixed

rachitnigam, 2023-01-01:
Fixed by https://github.com/cucapra/calyx/pull/1285

closed 2023-01-01 by rachitnigam