calyxir/calyx #1261: Deprecate `std_mem`

rachitnigam, 2022-11-13:
Standard memories are, in general, not synthesizable to BRAMs or URAMs on FPGAs because of combinational reads. The new sequential memories provide sequential, latency-insensitive interfaces to memories and are amenable to mapping optimizations like #1151.

In addition to #907, we should move the `std_mem` primitives into the `unsynthesizable.futil` file to signal that they cannot be easily synthesized and used

andrewb1999, 2022-11-15:
Before we do this we should add seq_mems to CIRCT and move passes over to use them instead.

rachitnigam, 2022-11-15:
@cgyurgyik thoughts on how hard this would be? The difference is that reads from memories require setting `read_en` signal and waiting till `read_done` is high before reading the value

cgyurgyik, 2022-11-15:
I imagine the brunt of it will come from refactoring. The PipelineToCalyx pass may be trickier, depending on how specialized memory accesses are. I don't recall exactly.

rachitnigam, 2023-01-01:
We can, for now, leave the `std_mem` in the repo when we do this and write a short pass to emit a warning whenever they are used by a frontend. We can eventually remove them completely once CIRCT stops using them

rachitnigam, 2023-01-02:
Linking a conversation that motivated this: https://github.com/cucapra/calyx/discussions/1221

cgyurgyik, 2023-07-19:
I think effort towards this was made on the CIRCT side in https://github.com/llvm/circt/pull/5471.

rachitnigam, 2024-02-06:
https://github.com/calyxir/calyx/pull/1901 renamed `std_mem_*` memories to `comb_mem_*` to indicate that they have combinational reads. I'm going to consider this sufficient to close this PR. In the future, we can consider removing these memories but for now, we should systematically eliminate their use from the code base.

closed 2024-02-06 by rachitnigam