calyxir/calyx #1277: Cannot infer static timing on AMC memory component

andrewb1999, 2022-11-18:
Here is a simplified version of a program that could be generated by AMC/CIRCT:
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
extern "amc.v" {
  primitive mem_0_prim[WIDTH, PORT0_SIZE, PORT0_IDX_SIZE, PORT1_SIZE, PORT1_IDX_SIZE](@clk clk: 1, port0_addr0: PORT0_IDX_SIZE, @go @static(1) port0_read_en: 1, port1_addr0: PORT1_IDX_SIZE, port1_write_data: WIDTH, @go(2) @static(1) port1_write_en: 1) -> (port0_read_data: WIDTH, @done port0_read_done: 1, @done(2) port1_write_done: 1);
}
component mem_0_comp(port1_addr0: 3, port1_write_data: 32, @go @static(1) port1_write_en: 1, @clk clk: 1, @reset reset: 1) -> (@done port1_write_done: 1) {
  cells {
    mem_0 = mem_0_prim(32, 8, 3, 8, 3);
  }
  wires {
    mem_0.clk = clk;
    mem_0.port1_write_data = port1_write_data;
    mem_0.port1_write_en = port1_write_en;
    port1_write_done = mem_0.port1_write_done;
    mem_0.port1_addr0 = port1_addr0;
  }
  control {
  }
}
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1) -> (@done done: 1) {
  cells {
    std_slice_5 = std_slice(32, 3);
    std_slice_4 = std_slice(32, 3);
    inst_1 = mem_0_comp();
    @external(1) mem_0 = std_mem_d1(32, 8, 3);
    while_0_arg0_reg = std_reg(32);
  }
  wires {
    group bb0_3 {
      std_slice_5.in = while_0_arg0_reg.out;
      std_slice_4.in = while_0_arg0_reg.out;
      inst_1.port1_addr0 = std_slice_5.out;
      inst_1.port1_write_data = mem_0.read_data;
      inst_1.port1_write_en = 1'd1;
      mem_0.addr0 = std_slice_4.out;
      bb0_3[done] = inst_1.port1_write_done;
    }
  }
  control {
    bb0_3;
  }
}
```
Given that `port1_write_en` is marked as `@static(1)`, I would expect `bb0_3` to be inferred to `@static(1)` as well. Instead Calyx is unable to infer any static value for that group. Not sure what the cause is here.

rachitnigam, 2022-11-18:
Will look into it. One workaround in the meantime is adding the `"static"` annotation on the group directly:
```
group bb0_3<"static"=1> { ... }
```

rachitnigam, 2022-11-18:
Simplified the program a bit more:
```
import "primitives/core.futil";
component mem_0_comp(@go @static(1) en: 1) -> (@done done: 1) {
  cells {}
  wires {}
  control {}
}
component main() -> () {
  cells {
    inst_1 = mem_0_comp();
  }
  wires {
    group bb0_3 {
      inst_1.en = 1'd1;
      bb0_3[done] = inst_1.done;
    }
  }
  control {
    bb0_3;
  }
}
```

rachitnigam, 2022-11-18:
Okay, the problem is that we're not correctly tracking information for component's latencies. I know how to fix it

PR #1279 (2022-11-19): Correctly infer static timing for Calyx components
- Add timing specified timing information to latency_data
- Fix #1277
commit d6977642 (2022-11-19): Fix #1277 (#1279)

* Add timing specified timing information to latency_data

* Fix #1277
closed 2022-11-19 by rachitnigam