calyxir/calyx #1278: Group2Invoke pass messes with static timing

andrewb1999, 2022-11-18:
With an example vadd program, group2invoke ends up significantly degrading the final latency of the design. Running this design with `-p pre-opt -d group2invoke` gives a top level static latency of 18, but running it with just `-p pre-opt` gives a top level static latency of 27. I'm not quite sure what optimizations are provided by group2invoke, but either there is a bug here or group2invoke should be turned off when using static timing.

Here's the design for reference:
```
import "primitives/core.futil";
component main<"toplevel"=1>(@clk clk: 1, @reset reset: 1, @go go: 1) -> (@done done: 1) {
  cells {
    std_slice_2 = std_slice(32, 3);
    std_slice_1 = std_slice(32, 3);
    std_slice_0 = std_slice(32, 3);
    std_add_1 = std_add(32);
    std_add_0 = std_add(32);
    std_lt_0 = std_lt(32);
    @external(1) mem_2 = std_mem_d1(32, 8, 3);
    @external(1) mem_1 = std_mem_d1(32, 8, 3);
    @external(1) mem_0 = std_mem_d1(32, 8, 3);
    stage_1_register_1_reg = std_reg(32);
    stage_1_register_0_reg = std_reg(32);
    stage_0_register_2_reg = std_reg(32);
    stage_0_register_1_reg = std_reg(32);
    stage_0_register_0_reg = std_reg(32);
    while_0_arg0_reg = std_reg(32);
  }
  wires {
    group bb0_0 {
      stage_0_register_2_reg.in = while_0_arg0_reg.out;
      stage_0_register_2_reg.write_en = 1'd1;
      bb0_0[done] = stage_0_register_2_reg.done;
    }
    group assign_while_0_init_0 {
      while_0_arg0_reg.in = 32'd0;
      while_0_arg0_reg.write_en = 1'd1;
      assign_while_0_init_0[done] = while_0_arg0_reg.done;
    }
    group bb0_3 {
      stage_1_register_1_reg.in = stage_0_register_2_reg.out;
      stage_1_register_1_reg.write_en = 1'd1;
      bb0_3[done] = stage_1_register_1_reg.done;
    }
    comb group bb0_6 {
      std_lt_0.left = while_0_arg0_reg.out;
      std_lt_0.right = 32'd8;
    }
    group bb0_7 {
      std_slice_2.in = while_0_arg0_reg.out;
      mem_0.addr0 = std_slice_2.out;
      stage_0_register_0_reg.in = mem_0.read_data;
      stage_0_register_0_reg.write_en = 1'd1;
      bb0_7[done] = stage_0_register_0_reg.done;
    }
    group bb0_8 {
      std_slice_1.in = while_0_arg0_reg.out;
      mem_1.addr0 = std_slice_1.out;
      stage_0_register_1_reg.in = mem_1.read_data;
      stage_0_register_1_reg.write_en = 1'd1;
      bb0_8[done] = stage_0_register_1_reg.done;
    }
    group bb0_9 {
      std_add_0.left = while_0_arg0_reg.out;
      std_add_0.right = 32'd1;
      while_0_arg0_reg.in = std_add_0.out;
      while_0_arg0_reg.write_en = 1'd1;
      bb0_9[done] = while_0_arg0_reg.done;
    }
    group bb0_10 {
      std_add_1.left = stage_0_register_0_reg.out;
      std_add_1.right = stage_0_register_1_reg.out;
      stage_1_register_0_reg.in = std_add_1.out;
      stage_1_register_0_reg.write_en = 1'd1;
      bb0_10[done] = stage_1_register_0_reg.done;
    }
    group bb0_11 {
      std_slice_0.in = stage_1_register_1_reg.out;
      mem_2.addr0 = std_slice_0.out;
      mem_2.write_data = stage_1_register_0_reg.out;
      mem_2.write_en = 1'd1;
      bb0_11[done] = mem_2.done;
    }
  }
  control {
    seq {
      assign_while_0_init_0;
      par {
        bb0_7;
        bb0_8;
        bb0_0;
        bb0_9;
      }
      par {
        bb0_7;
        bb0_8;
        bb0_0;
        bb0_9;
        bb0_10;
        bb0_3;
      }
      @bound(6) while std_lt_0.out with bb0_6 {
        par {
          bb0_7;
          bb0_8;
          bb0_0;
          bb0_9;
          bb0_10;
          bb0_3;
          bb0_11;
        }
      }
      par {
        bb0_10;
        bb0_3;
        bb0_11;
      }
      bb0_11;
    }
  }
}
```

rachitnigam, 2022-11-18:
This is related to #911: the problem isn't that `group2invoke` does anything bad, it's that `remove-comb-group` swoops in and converts `invoke x(...) with cg` to `seq { cg_to_g; invoke x(...) }` which increases latency for each `invoke-with`. Once #911 is solved, this should be fixed

andrewb1999, 2022-11-18:
Oh ok, great! I haven't looked much at the invoke syntax so I didn't know it also uses comb-groups.

rachitnigam, 2022-11-18:
Yup! https://docs.calyxir.org/lang/ref.html#invoke

rachitnigam, 2022-11-21:
@andrewb1999 if the above explanation makes sense, let's close this issue because fixing #911 will also fix this

andrewb1999, 2022-11-21:
Yep, that's fine with me. I'll go ahead and close this.

closed 2022-11-21 by andrewb1999