calyxir/calyx #1283: `-x inline:always` giving error

calebmkim, 2022-11-21:
Always inlining gives an error in the following Calyx design: 

[l.futil.txt](https://github.com/cucapra/calyx/files/10052683/l.futil.txt)

Basically, there is a component, `fp_pow`, which itself contains an invoke of a register. That component is read from in the main, which is giving an error when I try to fully inline. I've been trying to figure out what's wrong w/ it but if anyone else that understands the `component-inliner` pass better could give their input that would help.

rachitnigam, 2022-11-21:
What’s the error message?

rachitnigam, 2022-11-21:
The invoke to the register must be compiler into a group since primitive invoked cannot be compiled. I’d recommend disabling group2invoke

calebmkim, 2022-11-21:
sounds good that's what I've been doing. (Also, I've updated the original issue comment to include the error message)

rachitnigam, 2022-11-21:
Run with backtrace to see the stack trace

calebmkim, 2022-11-21:
The error is happening in the `rewrite_invoke()`  function. Specifically, it occurs when we are going through the inputs of the `invoke` and we are trying to get the parent name of one of the input ports, that's when it looks like the error is happening.

rachitnigam, 2022-11-21:
Can you try looking to which port is the problem? One option is to use VS Code's debugger to step through the program. You can use the "Debug Compiler" option to provide the arguments and the file and step through the program to see where the bug is coming from

rachitnigam, 2022-11-21:
> The invoke to the register must be compiler into a group since primitive invoked cannot be compiled. I’d recommend disabling group2invoke

Also, to be clear, this is problem with how `-x inline:always` works and interacts with `group2invoke`. We should expose some option like `-x group2invoke:skip-primitives` to ensure that it doesn't turn primitive groups to invokes

sampsyo, 2022-11-21:
Just tacking on an additional probably-wrong thought, just in case: instead of telling `group2invoke` not to deal with primitives, would an alternative be to tell the `inline:always` mode to not actually attempt to inline primitives?

rachitnigam, 2022-11-22:
The version of the program with only components doesn't error out:
```
import "primitives/core.futil";

component foo() ->  (out: 32) {
  cells {}
  wires {}
  control {}
}
component bar(in: 32) ->  (out: 32) {
  cells {}
  wires {}
  control {}
}
component main(@go go: 1, @clk clk: 1, @reset reset: 1) -> (@done done: 1) {
  cells {
    c = foo();
    b = bar();
  }
  wires {
  }
  control {
    seq {
      invoke b(in = c.out)();
    }
  }
}
```

rachitnigam, 2022-11-22:
I looked at the error and it seems to occur during the final program printing and *not the pass itself*. This means that the likely problem is that the inlining pass is not rewriting the assignment at all and when the pretty printer attempt to print `c.out`, it realizes that `c` is no longer defined

closed 2023-01-02 by rachitnigam