calyxir/calyx #1284: Large Calyx Designs overflow when running Calyx Compiler (before lowering to verilog)

calebmkim, 2022-11-21:
#1280 helped to solve the overflow for some of the larger Calyx designs. However, for the 2 largest (mobile and googlenet), I still have the problem of: 
```
thread 'main' has overflowed its stack
fatal runtime error: stack overflow
Aborted (core dumped)
```
However, this fails when I just run the _compiler_ (i.e., even when I don't try to compile to Verilog). 
I checked, and all of the passes run without overflow. This leads me to believe it's probably an error that occurs when we print the fully compiled Calyx program. 

[mobile.futil.txt](https://github.com/cucapra/calyx/files/10051386/mobile.futil.txt)
This is the mobilenet futil file. One thing is that to always inline you must disable the `group2invoke` pass first b/c of #1283.

rachitnigam, 2022-11-21:
You can run the compiler with `--log info` to see exactly which part of the compilation pipeline is failing. It also attempts to provide timing information. Also, if the compiler is erroring out, how were you generating synthesis information for these? Does it only overflow with `-x inline:always`? 

I would encourage you to add the exact command line invocation you're running in issues like these to ensure we can reproduce the problem

rachitnigam, 2022-11-21:
```
cargo run --release -- -d group2invoke -x inline:always -b verilog --log info mobile.futil > /dev/null
```

Generates the code correctly for me. However, I need to change default passes:
```patch
diff --git a/calyx/src/default_passes.rs b/calyx/src/default_passes.rs
index db32b70c..ed679898 100644
--- a/calyx/src/default_passes.rs
+++ b/calyx/src/default_passes.rs
@@ -76,9 +76,9 @@ impl PassManager {
                 CompileSync,
                 GroupToSeq,
                 GroupToInvoke, // Creates Dead Groups potentially
+                CompileRef,    //Must run before cell-share.
                 ComponentInliner,
                 CombProp,
-                CompileRef, //Must run before cell-share.
                 InferShare,
                 CellShare, // LiveRangeAnalaysis should handle comb groups
                 RemoveCombGroups, // Must run before infer-static-timing
```

We should probably make this happen by default since the inliner cannot inline programs with `ref` cells

calebmkim, 2022-11-21:
Switching the order of the CompileRef passes is in the PR for the CDF generator. 

That command works for me as well. What I was doing before was moving the GroupToInvoke pass to execute after ComponentInliner, and then keeping the `group2invoke`. For some reason that gives the `stack overflow` error. 

So running:  `cargo run -- -x inline:always --log info mobile.futil > /dev/null` gives the stack overflow error (with the `GroupToInvoke` pass moved to just after `ComponentInliner` Pass). 

Either way, I guess what we could do for right now is just disable group2invoke. That will solve this problem and #1283. 

Here's what the error looks like: 
[err.txt](https://github.com/cucapra/calyx/files/10053517/err.txt)

rachitnigam, 2023-04-29:
I can't seem to trigger the error. I tried moving group2invoke before inlining but that seems to work fine. Closing for now but we can reopen you can reproduce it @calebmkim

closed 2023-04-29 by rachitnigam