calyxir/calyx #1294: Watchpoint doesn't seem to work

rachitnigam, 2022-12-01:
I tried to add watch points to the following program in the debugger but it didn't work.
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component main() -> () {
  cells {
    @external foo_b0 = std_mem_d1(32, 2, 32);
    @external foo_b1 = std_mem_d1(32, 2, 32);
    @external baz_b0 = std_mem_d1(32, 2, 32);
    @external baz_b1 = std_mem_d1(32, 2, 32);
    idx_b0_0 = std_reg(32);
    incr_b0_0 = std_add(32);
    lt_b0_0 = std_lt(32);
    add_b0_0 = std_add(32);
    idx_b1_0 = std_reg(32);
    incr_b1_0 = std_add(32);
    lt_b1_0 = std_lt(32);
    add_b1_0 = std_add(32);
  }
  wires {
    group incr_idx_b0_0 {
      incr_b0_0.left = idx_b0_0.out;
      incr_b0_0.right = 32'd1;
      idx_b0_0.in = incr_b0_0.out;
      idx_b0_0.write_en = 1'd1;
      incr_idx_b0_0[done] = idx_b0_0.done;
    }
    comb group cond_b0_0 {
      lt_b0_0.left = idx_b0_0.out;
      lt_b0_0.right = 32'd2;
    }
    group eval_body_b0_0 {
      foo_b0.addr0 = idx_b0_0.out;
      baz_b0.addr0 = idx_b0_0.out;
      add_b0_0.left = foo_b0.read_data;
      add_b0_0.right = 32'd5;
      baz_b0.write_data = add_b0_0.out;
      baz_b0.write_en = 1'd1;
      eval_body_b0_0[done] = baz_b0.done;
    }
    group incr_idx_b1_0 {
      incr_b1_0.left = idx_b1_0.out;
      incr_b1_0.right = 32'd1;
      idx_b1_0.in = incr_b1_0.out;
      idx_b1_0.write_en = 1'd1;
      incr_idx_b1_0[done] = idx_b1_0.done;
    }
    comb group cond_b1_0 {
      lt_b1_0.left = idx_b1_0.out;
      lt_b1_0.right = 32'd1;
    }
    group eval_body_b1_0 {
      foo_b1.addr0 = idx_b1_0.out;
      baz_b1.addr0 = idx_b1_0.out;
      add_b1_0.left = foo_b1.read_data;
      add_b1_0.right = 32'd5;
      baz_b1.write_data = add_b1_0.out;
      baz_b1.write_en = 1'd1;
      eval_body_b1_0[done] = baz_b1.done;
    }
  }
  control {
    seq {
      par {
        while lt_b0_0.out with cond_b0_0 {
          seq {
            eval_body_b0_0;
            incr_idx_b0_0;
          }
        }
        while lt_b1_0.out with cond_b1_0 {
          seq {
            eval_body_b1_0;
            incr_idx_b1_0;
          }
        }
      }
    }
  }
}
```

In the debugger CLI:
```
 > watch after main::eval_body_b1_0 with print-state \u baz_b1
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
 > s
Main component has finished executing. Debugger is now in inspection mode.
```

I would've expected it to print the value in `baz_b1` once during the single steps

cgyurgyik, 2022-12-01:
Is `idx_b1_0`'s value ever set before `cond_b1_0` is run?

rachitnigam, 2022-12-02:
I definitely would expect a watch point to trigger when single stepping

EclecticGriffin, 2023-05-01:
Closing as I don't intend to change this in the immediate future. For the moment, this is expected behavior

closed 2023-05-01 by EclecticGriffin