calyxir/calyx #1303: Combinational logic generated when `[done]` hole comes from a `std_wire`

sampsyo, 2022-12-06:
Here is an interesting bug that I stumbled across by accident. The issue seems to require _both_ a `group[done]` hole being driven by a `std_wire` component _and_ control flow wrapping the group enable, which means that the likely culprit is control compilation.

Here is a reduced test case:

```
import "primitives/core.futil";
component main(flag: 1) -> () {
  cells {
    r = std_reg(32);
    w = std_wire(1);
  }
  wires {
    group g {
      r.write_en = 1'd1;
      r.in = 32'd5;
      w.in = r.done;
      g[done] = w.out;
    }
  }
  control {
    if flag {
      g;
    }
  }
}
```

Notably, a slightly tweaked version of the program that does _not_ use a `std_wire` indirection works fine:

```
import "primitives/core.futil";
component main(flag: 1) -> () {
  cells {
    r = std_reg(32);
    w = std_wire(1);
  }
  wires {
    group g {
      r.write_en = 1'd1;
      r.in = 32'd5;
      g[done] = r.done;
    }
  }
  control {
    if flag {
      g;
    }
  }
}
```

In either case, here is a command to run the program:

```
fud e -vv red.futil --to dat --through verilog -s verilog.data empty.json
```

There are no memories involved, so the input data file `empty.json` can just contain `{}`.

Verilator complains of a combinational cycle:

```
[fud] DEBUG: verilator -cc --trace /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad --exe /Users/asampson/cu/research/futil/fud/sim/testbench.cpp --build --top-module main --Mdir /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpkmzpv1if>&2
[fud] DEBUG: Stdin is `<class 'NoneType'>`
%Warning-UNOPTFLAT: /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:490:7: Signal unoptimizable: Circular combinational logic: 'main.w_in'
  490 | logic w_in;
      |       ^~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=5.002
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:490:7:      Example path: main.w_in
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:579:18:      Example path: ASSIGNW
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:500:7:      Example path: main.g_done_in
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:580:16:      Example path: ASSIGNW
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:498:7:      Example path: main.g_go_in
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:588:13:      Example path: ASSIGNW
                    /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdmbuv3ad:490:7:      Example path: main.w_in
```

Some light can be shed by looking at the lowered Calyx code (`fud e -vv red.futil --to futil-lowered`). These lines are cyclic:

```
g_go.in = !g_done.out & fsm.out == 2'd1 & tdcc_go.out ? 1'd1;
w.in = g_go.out ? r.done;
g_done.in = w.out;
```

The `done` signal guards the `go` signal as usual, and a compilation pass is inserting the condition`g_go.out ?` into the `w.in` assignment. I don't (yet) know which pass is responsible, but it seems like the problem is the unnecessary `go`-gating of combinational logic. (It only seems to matter that stateful logic is `go`-gated.)

rachitnigam, 2022-12-07:
We could also require that a group's done condition can only ever use specially marked `@done` ports. Also, the above reasoning can explain why something like `g[done] = r1.done & r2.done` is invalid because it'll roughly desugar to:
```
group g {
  and.left = r1.done;
  and.right = r2.done;
  g[done] = and.out;
}
```
Which is similarly problematic

Of course, we currently *have* such done conditions and those don't generate the above warning and that's because `r1.done & r2.done` is actually a continuous assignment that looks like this:
```
// assignment to and happen outside the group
and.left = r1.done;
and.right = r2.done;
group g {
  g[done] = and.out;
}
```

rachitnigam, 2022-12-10:
I’m not sure if we can do the latter unless no other group uses that component. The CIRCT code base has a “group invariant code motion” pass that turns group assignments into continuous assignments in specific cases like this (although it is not very fleshed out). The upside is that the generated hardware is simpler but one the other hand, you lose out on sharing opportunities. We should investigate something like this for combinational components, however, because it would be a good trade-off.

sampsyo, 2022-12-10:
Yep, good idea. That "unconditional stuff in groups" thing from above is pretty clearly a bad idea on second thought!

rachitnigam, 2023-01-01:
Updated the issue to reflect the actionable TODO item: Implement a papercut pass to warn when there is a combinational path between the done condition and one of the assignment sources. This is not going to be complete in general but hopefully still useful

sampsyo, 2023-01-03:
Yes, sounds good. Maybe we should add something about this to [the UB page in the docs](https://docs.calyxir.org/lang/undefined.html)? TBH I am not sure at the moment how to phrase the general restriction…

sampsyo, 2023-01-04:
I found myself thinking about something interesting that @rachitnigam hit on during a synchronous meeting: the intriguing fact that *guards are unshareable*. That's one of the things that makes guards special, with respect to other stuff in the Calyx language.

Further parroting what @rachitnigam pointed out, imagine that you have a group like this:

```
group g {
  a.in = b.out & c.out ? d.out;
}
```

(Please forgive the lack of a `done` condition; it doesn't matter in this line of argument.) Our compiler currently does not try to do any of its own logic synthesis (i.e., logical-to-structural translation) on guards. So something it does _not_ do, but which you could imagine it doing, would be to replace the `&` in the guard with an explicit `std_and` cell:

```
my_and = std_and(1);
...
group g {
  my_and.left = b.out;
  my_and.right = c.out;
  a.in = my_and.out ? d.out;
}
```

(Instead, the current compiler works by simply emitting the guard expression as a Verilog expression, letting the RTL backend can do its own combinational logic synthesis. As an aside, this is a nice division of labor so we don't have to jump into the logic-synthesis fray.)

One might assume that structure-ifying these guard expressions would work the same way whether it happens before or after compiling away groups. The reason I think this is interesting is that compiling the above, structure-ified version of the group would add _further_ `go` conditions to the program to implement isolation of the `g` group. Something like this:

```
my_and.left = g_go ? b.out;
my_and.right = g_go ? c.out;
a.in = g_go & my_and.out ? d.out;
```

The interesting thing is that the the first two conditions (on the first two lines) are *guaranteed to be useless*. `my_and`, because it comes from a guard, *may never be shared*, so these lines in `g` are certain to be the *only* assignments into its input ports, so we don't need to do any guarding of them at all. The weird thing is that Calyx IL has no way to express that: it doesn't have a way to tell the compiler, "hey, `my_and` here has a deep spiritual connection to `g`; no one else can ever use it. So don't bother guarding the inputs when you compile `g`." So I guess we can't implement that hypothetical guard-flattening pass without incurring a bunch of needless extra guarding downstream.

"Aha," I hear you saying. "There is *kinda* a way to express unshareability in Calyx, and to write assignments that don't get guarded with `go` signals during flattening! It's our favorite concept in the whole language: *continuous assignments*." You can imagine compiling the original `g` to this instead:

```
my_and = std_and(1);
...
my_and.left = b.out;
my_and.right = c.out;
group g {
  a.in = my_and.out ? d.out;
}
```

That is, assign to the components that implement guard expressions, like `my_and`, *continuously* and those won't get guarded. Great!

While it is kinda sad that we have lost any information about how `g` is spiritually associated with `my_and`, I think this is at least correct, *as long as we are super careful about undefinedness*. That is, the original program had the obligation to make `b.out & c.out` at least whenever `g` was active. I think we have been saying that this requires *both* `b.out` and `c.out` to be defined for the same time intervals. As long as `std_and` follows the same (or stricter) definedness rules as `&`, therefore, the compiled version of this program correctly makes `my_and.out` defined for the entire duration of `g`'s activation.

So it's possible that continuous assignments are the answer to our need for unshareable components and we should just be using those. Or perhaps we have a reason to have something that "compiles like" continuous assignments, in that they're unshareable, but also conveys that the components involved must *only* ever used within a certain group—which would enable a compiler pass to *add* sharing, for example, where it didn't previously exist.

This is all just to say that the need for unshareable logic is interesting, and I'm not sure either way whether we want new language constructs to support it.

This is probably also relevant to the trouble with #1326.

rachitnigam, 2023-01-05:
Thanks for the awesome summary @sampsyo! I think you got everything exactly right about the problem.

> I think this is at least correct, as long as we are super careful about undefinedness

The frustrating about this solution is that it reduces the compiler's "optimization power" because continuous assignments are global and increase the live-range of everything mentioned in them. An equivalent in software-land would be implementing all functional-local state by hoisting it into the global scope. While semantically correct (assuming your transformation of local-to-global is correct), it now disables a bunch of optimizations which is unfortunate.

One proposal I had is a notion of "complete assignments":
```
a.in = x ? 1 : 0;
```
Complete assignments indicate to the compiler that the assignment is not "open" in the same way guarded assignments are and that the port can only take the values mentioned in the assignment. Somehow, it should also indicate to the compiler that this means that the assignment should not be guarded in any manner when the group is lowered.

sampsyo, 2023-01-06:
Yes, exactly—the problem with the "continuous approach" above is that it loses the "containment" of the original logic. I don't yet have a concrete example of an optimization/etc. that is prohibited by this transformation, but it is not hard to buy that such an unfortunate limitation exists.

The only remedy is to invent something that combines the best of both worlds between continuous assignments and grouped assignments (where "best" is "most exploitable by the compiler"):

1. Don't need to be `go`-guarded because they must exclusively own their destinations, like continuous assignments.
2. Are guaranteed to only matter when the group is active, like grouped assignments.

These "complete assignments" are a cool idea for asserting exclusivity. But as you allude to at the end of your comment, I'm not quite sure how they provide benefit 1. That is, I imagine the idea is that they are still grouped, so we get benefit 2:

```
group g {
  a.in = x ? 1 : 0;
}
```

However, at least in my first guess at understanding the semantics, this assignment (like an ordinary assignment) only does anything when `g` is active. So it should also be legal to write a program with two complete assignments to the same port, as long as they are in different groups that are never active in parallel with one another:

```
group g {
  a.in = x ? 1 : 0;
}
group h {
  a.in = y ? 1 : 0;
}
...
seq { g; h; }
```

That in turn implies that these assignments must still be `go`-guarded when compiled, lest they conflict with each other post-compilation. Something like this:

```
a.in = g_go & x ? 1;
a.in = g_go & !x ? 0;
a.in = h_go & y ? 1;
a.in = h_go & !y ? 0;
```

…and we're sort of back where we started. Put differently, at least the way I'm interpreting them, "complete assignments" are only complete w/r/t they're explicit guards—not the implicit guarding from the group they appear in. They would be useful in themselves (I frequently find myself writing complementary assignment pairs that would be better expressed as complete assignments), but they don't seem to by themselves solve the `go`-guarding problem.