calyxir/calyx #1304: Conditional `reg.in` assignments are made unconditional during compilation

sampsyo, 2022-12-06:
I also recently stumbled across this surprisingly-simple program that exhibits an interesting compilation bug:

```
import "primitives/core.futil";
component main() -> () {
  cells {
    r = std_reg(32);
  }
  wires {
    group g {
      r.write_en = 1'd1;
      r.in = 1'd0 ? 32'd1;
      r.in = 1'd0 ? 32'd2;
      g[done] = r.done;
    }
  }
  control {
    g;
  }
}
```

The issue is those two conditional assignments to `r.in`. Notice that the conditions are both constant zeroes, so _neither_ assignment should fire. After compilation, they _both_ fire, leading to a conflict (that Verilator catches, gratefully).

Notably, adding a layer of indirection through a `std_wire` avoids the problem:

```
import "primitives/core.futil";
component main() -> () {
  cells {
    r = std_reg(32);
    w = std_wire(32);
  }
  wires {
    group g {
      w.in = 1'd0 ? 32'd1;
      w.in = 1'd0 ? 32'd2;

      r.write_en = 1'd1;
      r.in = w.out;
      g[done] = r.done;
    }
  }
  control {
    g;
  }
}
```

Notice that this version does the same thing, but the conditional assignments are instead to `w.in`, and then we copy that to `r.in`. So something seems to be sensitive specifically to conditional assignments _into registers_.

Here's how to run this program:

```
fud e -vv red.futil --to dat --through verilog -s verilog.data empty.json
```

Again, the input doesn't matter, so `empty.json` may just contain `{}`.

Verilator's error message is straightforward:

```
[Verilator] Simulation begin
[1] %Error: tmpdejjl255:676: Assertion failed in TOP.main: Multiple assignment to port `r.in'.
%Error: /var/folders/dj/jj7bq9q570908ggkt0vcfvch0000gn/T/tmpdejjl255:676: Verilog $stop
```

Diving into the lowered Calyx code, `fud e -vv red.futil --to futil-lowered`, it's easy to spot the problem:

```
r.in = invoke0_go.out ? comb_reg.out;
r.in = invoke0_go.out ? comb_reg0.out;
```

Those are _unconditional_ assignments into the register—except for the `go` guard. That's undoubtedly a problem. But what are `comb_reg*`? Here are the relevant assignments for one of them:

```
comb_reg.in = comb_invoke0_go.out ? in_guarded_wire.out;
comb_reg.write_en = comb_invoke0_go.out ? 1'd1;
in_guarded_wire.in = 1'd0 & comb_invoke0_go.out ? 32'd2;
```

The trouble here is that `r` is getting kinda "double-buffered," in the sense that it becomes two registers after compilation. Data goes to `comb_reg` first, and then it goes into `r`. The assignment to `comb_reg.in` is correctly conditioned! But the compiler thought it could just propagate `comb_reg.out` into `r.in`, which doesn't work because, unlike the RHS of the original assignment, *`comb_reg.out` is always defined*. Undefinedness strikes again!!

The wires in question arise in `remove-comb-groups`, which surely indicates that `group2invoke` is turning my hand-crafted register assignment into an `invoke`. Indeed, running `futil -p group2invoke red.futil` reveals:

```
component main(@go go: 1, @clk clk: 1, @reset reset: 1) -> (@done done: 1) {
  cells {
    r = std_reg(32);
    @generated in_guarded_wire = std_wire(32);
    @generated in_guarded_wire0 = std_wire(32);
  }
  wires {
    [...]
    comb group comb_invoke {
      in_guarded_wire.in = 1'd0 ? 32'd1;
      in_guarded_wire0.in = 1'd0 ? 32'd2;
    }
  }

  control {
    invoke r(
      in = in_guarded_wire.out,
      in = in_guarded_wire0.out
    )() with comb_invoke;
  }
}
```

Check out those multiple assignments to `in`! So to be clear, the fault lies with `group2invoke`, not with `remove-comb-groups` at all. `group2invoke` incorrectly assumes it's OK to take _any_ assignment RHS, conditional or no, and stick it into a wire, and the same value will then come out of the other end of the wire. That is, the pass assumes that this:

```
port = <whatever>;
```

Is equivalent to:

```
my_great_new_wire.in = <whatever>;
port = my_great_new_wire.out;
```

Not so! That's fine _only if_ `<whatever>` does not contain `?`. Undefinedness does not propagate through a `std_wire`.

I think this is surprisingly tricky and has important implications for the broader undefinedness discussion (#922). Namely, it really highlights the difference between the semantics of "a port that has not been assigned to yet" and "an unknown value that should not be trusted." The latter is more like Verilog's X, whereas the former is more Calyx-specific.

rachitnigam, 2022-12-10:
@calebmkim can you take a look at this when you get a chance?

rachitnigam, 2022-12-23:
So I had a chance to look at this more carefully after reviewing #1325 and I think the diagnosis of the problem isn't quite right. Running `group2invoke` generates the following:
```
import "primitives/core.futil";
component main(@go go: 1, @clk clk: 1, @reset reset: 1) -> (@done done: 1) {
  cells {
    r = std_reg(32);
    @generated in_guarded_wire = std_wire(32);
    @generated in_guarded_wire0 = std_wire(32);
  }
  wires {
    group g {
      r.write_en = 1'd1;
      r.in = 1'd0 ? 32'd1;
      r.in = 1'd0 ? 32'd2;
      g[done] = r.done;
    }
    comb group comb_invoke {
      in_guarded_wire.in = 1'd0 ? 32'd1;
      in_guarded_wire0.in = 1'd0 ? 32'd2;
    }
  }

  control {
    invoke r(
      in = in_guarded_wire.out,
      in = in_guarded_wire0.out
    )() with comb_invoke;
  }
}
```

The problem is that we're generating a wire for _each guarded assignment_ to the port but instead we should be generating _exactly one wire_ for a port and then providing all the guarded assignments into that wire. That is, the code should look like:
```
    comb group comb_invoke {
      in_guarded_wire.in = 1'd0 ? 32'd1;
      in_guarded_wire.in = 1'd0 ? 32'd2;
    }
  }

  control {
    invoke r(
      in = in_guarded_wire.out,
    )() with comb_invoke;
  }
```

@calebmkim does that make sense? I don't think we need to disable this transformation and instead make sure we're using the same wire for each guarded assignment to a port.

closed 2022-12-28 by rachitnigam