calyxir/calyx #1341: Handling `group[done] = reg.done` in LiveRangeAnalysis

calebmkim, 2023-01-12:
Consider the following group: 
```
group upd360<"static"=2> {
      mem.read_en = 1'd1;
      slice.in = ...;
      mem.addr0 = slice.out;
      r.write_en = m.read_done;
      r.in = mem.out;
      upd360[done] = r.done;
    }
```
Since `r` is technically being read from in the group (namely `r.done`), r will be unnecessarily live. This significantly decreases sharing opportunities.  

We have two options: 

1) Because of the `slice` we don't transform this group in the group2seq pass (group2seq can't handle groups that have assignments to combinational cells).  We could try to get `group2seq` to be able to split groups even when  there are assignments to combination cells.  If we are able to do this, then the resulting group (i.e., the second group after we split `upd360`) will be "variable like"--a special case in Live Range Analysis which ignores the read of `r` in `upd360[done] = r.done`. 

2) Add a method in `ReadWriteSet` that does the following: "get all the reads of ports, except for reads in the situation where we have `group[done] = _.done`. " This options seems easier. Is there a reason why this wouldn't work? 

Also, lmk if I'm not being clear.

rachitnigam, 2023-01-13:
I think overall this does make sense and in fact why we have `VariableLike` and `group2invoke` which ignore the `done` signal. Do you think it makes sense to always ignore any signal marked as `@done` from the use list? Or perhaps we want something stronger like "if both the `@go` and `@done` signal are used in the same group, then we can remove `@done` from the use list"

calebmkim, 2023-01-13:
Or, to be even more sure, something like: 
"if both the `@go` and `@done` signal are used in the same group, and the `@done` signal is from an assignment to the group's done signal, then we can remove the `@done` from the list". 

I'm not thinking of a specific counter example right now, but I was thinking: is it possible to have the `@done` signal be read from, and _then_, later on within the group, having the `@go` signal be triggered?

rachitnigam, 2023-01-14:
> "if both the @go and @done signal are used in the same group, and the @done signal is from an assignment to the group's done signal, then we can remove the @done from the list".

Yeah, but that's pretty specific and doesn't capture chaining-like cases:
```
group one {
  x.write_en = y.done;
  one[done] = x.done;
}
```

In this case, would we want to ignore both `x.done` and `y.done`?

> is it possible to have the @done signal be read from, and then, later on within the group, having the @go signal be triggered?

Groups can do whatever the want. In this case, a group could implement some sort of loop where the `@go` and `@done` signals are repeatedly triggered right? Was there a specific case you were thinking about?

calebmkim, 2023-01-16:
> In this case, would we want to ignore both x.done and y.done?

Yeah, you're right. What I had in my first try of the PR did not capture enough cases. I've updated it now. 

One thing I was thinking about was this. If we see 
`cell.go = !cell.done ? 1'd1` 
Can we ignore any `cell.done` as reads? I'm not thinking of a situation where we would need to keep it as live, but maybe I'm wrong?

calebmkim, 2023-01-28:
Addressed by https://github.com/cucapra/calyx/pull/1342

closed 2023-01-28 by calebmkim