calyxir/calyx #1354: Icarus Verilog and Verilog giving me Different Results. Does this reproduce for others?

calebmkim, 2023-02-07:
If I run: 
```
fud e tests/correctness/seq-mem-dot-product.futil \
  -s verilog.data tests/correctness/seq-mem-dot-product.futil.data \
  --to dat --through verilog -v
```
I get 220, in the "out" value, as expected. 

If I run (same thing but thru icarus-verilog not verilog): 
```
fud e tests/correctness/seq-mem-dot-product.futil \
  -s verilog.data tests/correctness/seq-mem-dot-product.futil.data \
  --to dat --through icarus-verilog -v
```
I get 0, in the "out" value, which is _not_ expected. 

Does this reproduce for other people? 

The `iverilog` version is 11.0, which is fine according to `fud check`. Another note is that I get a warning warning: 
```
[fud] WARNING: Unknown option `verilog.top_module' for stage verilog 
```
when I run thru verilog.

rachitnigam, 2023-02-07:
The warning is benign. Can you follow instruction for [debugging compilation bugs](https://docs.calyxir.org/debug/debug.html) to see if there is problem with any of the passes? If running with `-p no-opt` fixes the problem, then we know something is wrong in pass. Otherwise something is wrong in our Icarus test harness

calebmkim, 2023-02-07:
`-p no-opt` doesn't fix the error. 
 
One interesting thing that does fix the problem: in `memories.sv`, when we define `seq_mem_d1`, we have the following : 

```
  // Write value to the memory
  always_ff @(posedge clk) begin
    if (!reset && write_en)
      mem[addr0] <= write_data;
  end 
```
If we remove the `!reset`, and just have `if (write_en)` then that fixes the problem. The reason why this is notable is because the `!reset` part was recently added in the TDST PR that was just merged. 

So I think it's basically that the `!reset` condition is making it so that we're not writing to memory... but only for icarus-verilog, not for verilog.

rachitnigam, 2023-02-07:
Oh weird, that just says that you shouldn't write anything to the memory while `reset` is asserted. Can you look at the waveform and see what's going on in there?

calebmkim, 2023-02-07:
Sure; which wave viewer program should I use. Would just GTKWave (that's one of the ones listed in the documentation) be fine?

rachitnigam, 2023-02-07:
If you use a Mac, you can use Scanscion too

calebmkim, 2023-02-08:
Seems like we're just never triggering the `.reset` port of our `seq-mems`; the signal just says "x" for the entire time, and when I add a simple `out.reset = reset` to the Calyx file, it gives the correct/expected output. 

This is probably something for the TDST pass? Is there a specific place in the TDST code where I should look?

rachitnigam, 2023-02-08:
Huh, looks like I didn't. Can you look at the `reset-insertion` pass? It should add that assignment

calebmkim, 2023-02-09:
The pass skips cells with the `@external` attribute.

rachitnigam, 2023-02-09:
Ugh, that might have to do with https://github.com/cucapra/calyx/issues/1034? I think it should be fine to not ignore the `@external` cells anymore? This would also explain the problem I was running into with #1338

rachitnigam, 2023-02-16:
@calebmkim I might've missed this but did you open a PR to fix this?

calebmkim, 2023-02-16:
I didn't because I read #1034 and I wasn't quite sure I fully understood the issue. 

However, if all that's required is to make a PR to re-enable reset insertion for external cells, then I can definitely do that.

closed 2023-05-08 by rachitnigam