calyxir/calyx #1355: Missing documentation for combinational components

bcarlet, 2023-02-09:
I found out about these by accident while looking through past issues; it would be helpful to have them mentioned in the docs. `std_cat` is another recent addition that isn't documented.

rachitnigam, 2023-02-09:
Thanks for the pointer! Combinational components are relatively new feature so we haven't added documentation for it. In a nutshell, these components cannot have any `control` section or `group`s and are limited to just just continuous assignments. Their well-formedness constraint requires their logic to finish executing within a cycle.

Feel free to take a shot at adding this documentation if you'd like!

rachitnigam, 2023-04-27:
Closed by #1412