calyxir/calyx #1361: [Compiler] Expand keyword checks

EclecticGriffin, 2023-02-13:
The keyword checks need to be expanded to check components & primitives. The list itself may also need to be expanded.

This comes from the discussion here: https://github.com/cucapra/calyx/discussions/1356

In the future we could add keywords to the name generator so the names will automatically get adjusted.

rachitnigam, 2023-10-03:
Related issue: #1566

EclecticGriffin, 2024-05-06:
So current inspection suggests we never did this. It's probably still worth doing and I can knock it out if people would like. The only downside is that it might be confusing for people if the names get automatically adjusted rather than erroring out

EclecticGriffin, 2024-05-06:
Actually looks to be non-trivial since this touches a lot of places. Might be easier to expand the well-formedness check since not everything goes through the name generator?