calyxir/calyx #1363: Support `new_fsm` attribute for tdst pass

calebmkim, 2023-02-16:
In the same way that the `new_fsm` attribute prevents fsm complexity blow up in the tdcc pass, we might want to add support when compiling tdst as well. I can try to implement this, since it will probably be helpful for me to better understand tdst, especially if I'm trying to implement some of the new static group syntax (w/ the `%` relative clock cycles thing).

rachitnigam, 2023-02-16:
You can take a shot but the big problem is that the static FSMs don't reset correctly which means you can't compositionally compile the program. This is why our current implementation is so restrictive (one component, completely static).

The problem is that there is no good way (that I know of) to build a static FSM that provides both a `done` condition and resets correctly so that it can be re-executed in the same cycle. 

To observe this problem, you can run the [code mentioned in the fix-tdst PR](https://github.com/cucapra/calyx/pull/1338/files/43b82ca5b93b3acbbc78c97ac61f66ed091f7d8b..34591f06de061459151476281f271387c203841d#diff-fe21c333672a9de582767d63163deea8e1d2c9986ac44e65d3d7178d1b9e417b). Uncomment the code and delete the stuff before and after the loop. If you look at the generated VCD, you'll notice that the inner `if` branches don't reset in time to be re-executed every cycle.

calebmkim, 2023-02-16:
Ah I see. Maybe this is something that would be better to implement once the static group stuff is more developed.

rachitnigam, 2023-03-06:
I'm going to make this task blocked on the static timing implementation

rachitnigam, 2023-03-06:
Blocked on #1344

rachitnigam, 2023-06-27:
@calebmkim did we decide that this doesn't need to be solved or that it isn't obvious what it means in the context of the new static lowering pass?

calebmkim, 2023-06-27:
We never fully reached a conclusion, but rn I'm leaning towards it not being to be solved. 
I explain why I think that [here](https://github.com/cucapra/calyx/discussions/1523#discussioncomment-6136468). (although in the comment, you mention that it's possible that there may be some benefit to inlining static control w/ new fsms).

rachitnigam, 2023-06-27:
Right! Maybe we should mark this as a low priority QoR thing to investigate if we have the time?

calebmkim, 2024-05-13:
This discussion is subsumed by the overall FSM optimizations described in #2020.

closed 2024-05-13 by calebmkim