calyxir/calyx #1367: Use Par Timing Analysis to combine capabilities of StaticParConv and MergeStaticPar

calebmkim, 2023-02-17:
merge-static-par (msp) performs the following transformation:
```
par { @static(N) g1; @static(N) g2; @static(N) g3 }
```
into:
```
group G { ...g1, ...g2, ...g3 } // All assignments from the groups
par { @static(N) G }
```

static-par-conv (spc) performs the following transformation:
```
par {
    seq { @static(M) A0; @static(N) B0; @static(P) C0; }
    seq { @static(M) A1; @static(N) B1; @static(P) C1; }
    seq { @static(M) A2; @static(N) B2; @static(P) C2; }
}
```
into
```
@static(M + N + P) seq {
    @static(M) par{ @static(M) A0; @static(M) A1; @static(M) A2;}
    @static(N) par{ @static(N) B0; @static(N) B1; @static(N) B2;}
    @static(P) par{ @static(P) C0; @static(P) C1; @static(P) C2;}
}
```
### Idea

Basically, merge-static-par is using the fact that we can combine the assignments in groups that we know are going to be executed for exactly the same clock cycle. However, it only does the transformation for which each thread in the par is a single group. Instead, we can use par timing analysis to generalize merge static par, so that we can combine any groups in a static par that we know will be run exactly in parallel. 

So if we have: 
```
@static(X) par { seq{ (before A) @static(N) A (after A) }  seq{ (before B) @static(N) B (after B) }}
```
and we us par timing analysis to see that  A and B will be executed in exactly the same clock cycle interval, we can rewrite this as: 
```
@static(X) seq { par{before A; before B;}  @static A&B par{after A; after B;}}
```
where `A&B` includes the assignments of A and B combined. If you notice, this is really just a generalization of what static-par-conv does. So we're really combining msp and spc when we do this.

#1340 explains this is another way.

rachitnigam, 2023-02-18:
Another thing @paili0628 might be interested in is implementing a `@sync` conversion pass that uses the information from this analysis to transform dynamic `@sync` into time-based synchronization when possible

rachitnigam, 2023-05-11:
Subsumed by https://github.com/cucapra/calyx/issues/1464