calyxir/calyx #1373: Reduce size of ternary expressions generated by Verilog backend

priyasrikumar, 2023-02-22:
Calyx has been generating Verilog code that [way too big.](https://github.com/cucapra/calyx/issues/1360) The root cause of this, identified in Adrian's comment, is that assigns statements are amalgamated into one big ternary expression. We should break assignments into smaller chunks. (and maybe discuss the best way to do that here!)

sampsyo, 2023-02-23:
To state the obvious, the straightforward way to make this into separate lines is to use Verilog `if`:

```
if (msp_go_out) begin
  assign depth_output_addr0 = 13'd0
end else if (msp0_go_out) begin
  assign depth_output_addr0 = 13'd1
...
```

But I recall that we stopped doing that (and switched to the ternary expression style) for some reason… but I can't recall why. Maybe just testing that out to see if something goes obviously wrong would be informative?

rachitnigam, 2023-02-23:
I think there is was a problem with the semantics of `assign` and `always_comb ... if` not being exactly the same causing various problems. I think #1177 is the other approach we can consider.