calyxir/calyx #1374: Implementing `% time` and `static group` in ir

paili0628, 2023-02-22:
During meeting with @calebmkim, it occurred to us that we don't lose anything to have static timing signals 1. as expressions rather than ports 2. only be used in guards. To the contrary, it might be easier to generalize. For example, if we have these two `static group`s,
```
wires {
   static group write_1 {
     reg.in = 32'd5;
     reg.write_en = %0 ? 1'd1;
     write_1[done] = %1 ? 1'd1;
    }

  static group write_2 {
    reg.in = 32'd10;
    reg.write_en = %0 ? 1'd1;
    write_2[done] = %1 ? 1'd1;
  }
}

control {
   static seq {
      write_1;  write_2;
   }
}
```
What we would want to generate in Calyx in this case would be:

```
static group seq {
   reg.in = %0 ? 32'd5;
   reg.write_en = %0? 1'd1;
   reg.in = %1 ? 32'd10;
   reg.write_en = %1 ? 1'd1;
   seq[done] = %2 ? 1'd1;
}
```

In the code above, we see that `reg.write_en = %1` can be expressed as `reg.write_en = %1 ? 1'd1` without generating much more hardware; whereas for the line `reg.in = %1? 32'd10` we would need static timing to be used as a guard to express this information (at least without having to use a wire).
From the above example, we see that it is more convenient if we allow static timing signals (only) to be used in guards from the beginning, as this prevents the need for instantiating a `std_wire` that transports the static timing signal into a guard. This design has a few advantages:
1. At a higher level, the `%n` syntax is arguably more easily thought of as an expression that indicates the satisfaction of a condition rather than an input signal, which implicitly would mean "the 1 bit signal is up if and only if the relative clock is at this cycle".
2. Treating `StaticTiming` as a different struct than a `Port` in the ir and allowing the `Guard` struct to deal with it separately saves us from the ugliness of potentially having to change all the methods in `Port`. For example, we would define a new `Static Guard` Enum as follows: 
```
pub enum StaticGuard {
    /// Represents `c1 || c2`.
    Or(Box<Guard>, Box<Guard>),
    /// Represents `c1 && c2`.
    And(Box<Guard>, Box<Guard>),
    /// Represents `!c1`
    Not(Box<Guard>),
    /// The constant true
    True,
    /// Comparison operator.
    CompOp(PortComp, RRC<Port>, RRC<Port>),
    /// Uses the value on a port as the condition. Same as `p1 == true`
    Port(RRC<Port>),
    /// Adding Static Timing Signal 
    StaticTiming(...)
}
```
One challenge is that the behavior for things like `static_guard.all_ports()`, or `static_guard.for_each()` is not completely clear. But, we were thinking that if we compile static groups early enough in the compilation pipeline, we won't have to worry about the passes that use `guard.all_ports()` or other methods similar to it. Since most of the passes that currently use these sorts of methods are based on dynamic timing, hopefully this won't be too much of a problem?

rachitnigam, 2023-02-23:
Wow, this is a super cool insight! At first blush, this seems like a clear win. I was wondering if you’d ever want to use the relative time operation as a port to store its value but I can’t come up with a reason you’d want that.

I think one way to answer the question about the behavior of “all ports” is studying how we use it in the compiler. It might be safe to either not return the value of the relative time guard or explicitly panic if we ever see a static time guard when calling the method.

sampsyo, 2023-02-23:
Really good thinking here! It is easy to see the aesthetic advantages, at the very least: these static-timing markers "feel" much more akin to condition expressions, which are purely logical/behavioral, than they resemble ports, which are truly structural/physical. So it makes a lot of sense to think of them as part of that logical world.

One nit occurs to me—and, as always, it's related to undefinedness. There is no problem with the writes to `reg.in` in the example. But consider the writes to `reg.write_en`:

```
reg.write_en = %0 ? 1'd1;
```

And consider the difference w/r/t putting the time expression on the RHS:

```
reg.write_en = %0;
```

These two options are different:

* The first assigns 1 into `write_en` on cycle 0, and it assigns *nothing at all* into `write_en` on other cycles.
* The second assigns 1 into `write_en` on cycle 1, and it assigns 0 into `write_en` on other cycles.

I actually think that the added flexibility of the first option is useful. This would allow us to use the register twice within the same group, which was actually a conflict (which I did not notice previously!!!!) in our old design. That is, this code in the compiled group is fine:

```
reg.write_en = %0 ? 1'd1;
reg.write_en = %1 ? 1'd1;
```

But the "old version," where these expressions were on the RHS alone and both assignments were unconditional, would actually be simultaneously trying to assign 0 and 1 into the port at the same time.

However, it means that we need to return to thinking carefully about undefinedness. I think we do _not_ want to require inverted pairs of assignments, like this:

```
reg.write_en = %0 ? 1'd1;
reg.write_en = ~%0 ? 1'd0;
```

…because that creates the same conflict problems after compilation. So instead, we really need to somehow design our semantics such that undefinedness leads to disabling for control ports… but that itself is a can of worms. Some care is required here!

rachitnigam, 2023-02-23:
@sampsyo I’m going to get you an “undefinedness wizard” hat one day.

This is a really good point and I did not notice it either. Regarding the problem with “inverted assignments”, why can’t we rely on the standard backend’s property of always making 0 the default assignment for control ports?

sampsyo, 2023-02-24:
Looking forward to the hat!!!!!

Staying in the realm of semantics (as opposed to backend implementation choices): if we want to decide that the language semantics includes a sentence like this:

> Whenever a port would otherwise be undefined, and it is marked as an interface port, it instead gets the value 0.

…then yes, you do not need those problematic "inverted pairs," and compilation in the proposed world seems to work out OK. We have previously discussed that this semantic policy is _also_ kind of weird for its own reasons, however (it generates "ghost" zeroes that do not come from any explicit assignments, for instance), meaning that I would still like to nail down this semantic choice once and for all, preferably accompanied with a formal model. But I think that this proposal here does push us farther in that direction, and farther way from (for example) a policy where programs are _required to explicitly make interface ports always defined_, which is the "opposite" policy in essence.

rachitnigam, 2023-02-24:
Yeah, this discussion might also be relevant to @priyasrikumar & @hackedy

priyasrikumar, 2023-02-24:
Looks super cool! If I can ask for a clarification, can you break down this example for me? The second example seems strange-- I guess I realized I never thought of %time labels as assignable, and exclusively thought of them as guards (adding credibility to the intuitiveness of the proposal)! What does assigning a variable %0 do? Specifically here, why does it assign reg.write_en 1? and on cycle 1 (since I thought %0 would mean cycle 0)?

> One nit occurs to me—and, as always, it's related to undefinedness. There is no problem with the writes to reg.in in the example. But consider the writes to reg.write_en:
>
> `reg.write_en = %0 ? [...]

Related to the above, why might we have considered requiring inverted pairs of assignments? How would that nominally resolves the undefinedness issue (which, as I understand, is due to the current semantics permitting multiple assignments to ports)?

sampsyo, 2023-02-24:
For context, @paili0628 & @calebmkim's proposal here is a change from @rachitnigam's original proposal in #1344, where `%time` thingies appeared on the RHS of assignments instead of in the guard. So if you were expecting them to be in guards all along, then this change makes the idea align more with your intuition!

> Specifically here, why does it assign reg.write_en 1? and on cycle 1 (since I thought %0 would mean cycle 0)?

Totally a typo! Thanks. I have gone back and fixed it.

> Related to the above, why might we have considered requiring inverted pairs of assignments? How would that nominally resolves the undefinedness issue (which, as I understand, is due to the current semantics permitting multiple assignments to ports)?

The reason to use that funky "inverted pair" thing would be to ensure that `reg.write_en` is *always defined*. If, in our semantics, "undefined" means "we don't specify what value it takes, and the hardware can use any value at any time," then you can imagine that port wildly fluctuating between 0 and 1 whenever it's not being explicitly assigned to be one or the other. Which, for a register's `write_en` port, means that it will constantly be overwriting itself whenever the port happens to fluctuate to 1. That seems bad! So one way or another, we really want `reg.write_en` to be 0 at all points when it is not assigned to 1.

Undefinedness is a big and tricky issue (cf #922), but the status quo is that we're avoiding the problem because the backends are "defaulting" things to zero. As a brief summary, we also have concerns that this is an expensive thing to do and could be confusing in other ways. But it is of course much better than letting registers mysteriously overwrite themselves when no one is looking.

rachitnigam, 2023-04-12:
@calebmkim this is completed by https://github.com/cucapra/calyx/pull/1403 right?

calebmkim, 2023-04-12:
yes

closed 2023-04-12 by calebmkim