calyxir/calyx #1376: Only function with name `main` can be RTL synthesized.

HahaLan97, 2023-02-23:
I'm trying to use calyx as a hls back-end, which will process the `futil` file generated by `circt-opt` from `mlir`, but I got an error saying:
```
⠇ synth-verilog.run_vivado[fud] ERROR: Utilization implementation file /tmp/tmpgyftljvf/out/FutilBuild.runs/impl_1/main_utilization_placed.rpt is missing
[fud] ERROR: Timing file /tmp/tmpgyftljvf/out/FutilBuild.runs/impl_1/main_timing_summary_routed.rpt is missing
{
  "uram": -1,
  "cell_lut1": 6,
  "cell_lut2": 324,
  "cell_lut3": 103,
  "cell_lut4": 75,
  "cell_lut5": 38,
  "cell_lut6": 257,
  "cell_fdre": 300
}

```
The original mlir programm is:
```
module {
  func.func @matmul(%arg0: memref<9xi32>, %arg1: memref<9xi32>, %arg2: memref<9xi32>) {
    affine.for %arg3 = 0 to 3 {
      affine.for %arg4 = 0 to 3 {
        affine.for %arg5 = 0 to 3 {
          %0 = affine.load %arg0[%arg5 + %arg3 * 3] : memref<9xi32>
          %1 = affine.load %arg1[%arg5 + %arg4 * 3] : memref<9xi32>
          %2 = arith.muli %0, %1 : i32
          %3 = affine.load %arg2[%arg4 + %arg3 * 3] : memref<9xi32>
          %4 = arith.addi %3, %2 : i32
          affine.store %4, %arg2[%arg4 + %arg3 * 3] : memref<9xi32>
        }
      }
    }
    return
  }
}

```
Then I used this to generate the following native calyx: `mlir-opt matmul.mlir -lower-affine -scf-for-to-while | circt-opt -lower-scf-to-calyx | circt-translate -export-calyx -o matmul.futil`.
```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
component matmul<"toplevel"=1>(ext_mem0_read_data: 32, ext_mem0_done: 1, ext_mem1_read_data: 32, ext_mem1_done: 1, ext_mem2_read_data: 32, ext_mem2_done: 1, @clk clk: 1, @reset reset: 1, @go go: 1) -> (ext_mem0_write_data: 32, ext_mem0_addr0: 4, ext_mem0_write_en: 1, ext_mem1_write_data: 32, ext_mem1_addr0: 4, ext_mem1_write_en: 1, ext_mem2_write_data: 32, ext_mem2_addr0: 4, ext_mem2_write_en: 1, @done done: 1) {
  cells {
    std_slice_3 = std_slice(32, 4);
    std_slice_2 = std_slice(32, 4);
    std_slice_1 = std_slice(32, 4);
    std_slice_0 = std_slice(32, 4);
    std_add_7 = std_add(32);
    muli_4_reg = std_reg(32);
    std_mult_pipe_4 = std_mult_pipe(32);
    std_add_6 = std_add(32);
    load_0_reg = std_reg(32);
    std_add_5 = std_add(32);
    muli_3_reg = std_reg(32);
    std_mult_pipe_3 = std_mult_pipe(32);
    muli_2_reg = std_reg(32);
    std_mult_pipe_2 = std_mult_pipe(32);
    std_add_4 = std_add(32);
    muli_1_reg = std_reg(32);
    std_mult_pipe_1 = std_mult_pipe(32);
    std_add_3 = std_add(32);
    muli_0_reg = std_reg(32);
    std_mult_pipe_0 = std_mult_pipe(32);
    std_add_2 = std_add(32);
    std_slt_2 = std_slt(32);
    std_add_1 = std_add(32);
    std_slt_1 = std_slt(32);
    std_add_0 = std_add(32);
    std_slt_0 = std_slt(32);
    while_2_arg0_reg = std_reg(32);
    while_1_arg0_reg = std_reg(32);
    while_0_arg0_reg = std_reg(32);
  }
  wires {
    group assign_while_0_init_0 {
      while_0_arg0_reg.in = 32'd0;
      while_0_arg0_reg.write_en = 1'd1;
      assign_while_0_init_0[done] = while_0_arg0_reg.done;
    }
    group assign_while_1_init_0 {
      while_1_arg0_reg.in = 32'd0;
      while_1_arg0_reg.write_en = 1'd1;
      assign_while_1_init_0[done] = while_1_arg0_reg.done;
    }
    group assign_while_2_init_0 {
      while_2_arg0_reg.in = 32'd0;
      while_2_arg0_reg.write_en = 1'd1;
      assign_while_2_init_0[done] = while_2_arg0_reg.done;
    }
    comb group bb0_0 {
      std_slt_0.left = while_2_arg0_reg.out;
      std_slt_0.right = 32'd3;
    }
    comb group bb0_2 {
      std_slt_1.left = while_1_arg0_reg.out;
      std_slt_1.right = 32'd3;
    }
    comb group bb0_4 {
      std_slt_2.left = while_0_arg0_reg.out;
      std_slt_2.right = 32'd3;
    }
    group bb0_6 {
      std_mult_pipe_0.left = while_2_arg0_reg.out;
      std_mult_pipe_0.right = 32'd3;
      muli_0_reg.in = std_mult_pipe_0.out;
      muli_0_reg.write_en = std_mult_pipe_0.done;
      std_mult_pipe_0.go = 1'd1;
      bb0_6[done] = muli_0_reg.done;
    }
    group bb0_9 {
      std_mult_pipe_1.left = while_1_arg0_reg.out;
      std_mult_pipe_1.right = 32'd3;
      muli_1_reg.in = std_mult_pipe_1.out;
      muli_1_reg.write_en = std_mult_pipe_1.done;
      std_mult_pipe_1.go = 1'd1;
      bb0_9[done] = muli_1_reg.done;
    }
    group bb0_12 {
      std_slice_3.in = std_add_3.out;
      std_slice_2.in = std_add_4.out;
      std_mult_pipe_2.left = ext_mem0_read_data;
      std_mult_pipe_2.right = ext_mem1_read_data;
      muli_2_reg.in = std_mult_pipe_2.out;
      muli_2_reg.write_en = std_mult_pipe_2.done;
      std_mult_pipe_2.go = 1'd1;
      ext_mem0_addr0 = std_slice_3.out;
      std_add_3.left = while_0_arg0_reg.out;
      std_add_3.right = std_mult_pipe_0.out;
      ext_mem1_addr0 = std_slice_2.out;
      std_add_4.left = while_0_arg0_reg.out;
      std_add_4.right = std_mult_pipe_1.out;
      bb0_12[done] = muli_2_reg.done;
    }
    group bb0_13 {
      std_mult_pipe_3.left = while_2_arg0_reg.out;
      std_mult_pipe_3.right = 32'd3;
      muli_3_reg.in = std_mult_pipe_3.out;
      muli_3_reg.write_en = std_mult_pipe_3.done;
      std_mult_pipe_3.go = 1'd1;
      bb0_13[done] = muli_3_reg.done;
    }
    group bb0_15 {
      std_slice_1.in = std_add_5.out;
      ext_mem2_addr0 = std_slice_1.out;
      load_0_reg.in = ext_mem2_read_data;
      load_0_reg.write_en = 1'd1;
      std_add_5.left = while_1_arg0_reg.out;
      std_add_5.right = std_mult_pipe_3.out;
      bb0_15[done] = load_0_reg.done;
    }
    group bb0_17 {
      std_mult_pipe_4.left = while_2_arg0_reg.out;
      std_mult_pipe_4.right = 32'd3;
      muli_4_reg.in = std_mult_pipe_4.out;
      muli_4_reg.write_en = std_mult_pipe_4.done;
      std_mult_pipe_4.go = 1'd1;
      bb0_17[done] = muli_4_reg.done;
    }
    group bb0_19 {
      std_slice_0.in = std_add_7.out;
      ext_mem2_addr0 = std_slice_0.out;
      ext_mem2_write_data = std_add_6.out;
      ext_mem2_write_en = 1'd1;
      std_add_7.left = while_1_arg0_reg.out;
      std_add_7.right = std_mult_pipe_4.out;
      std_add_6.left = load_0_reg.out;
      std_add_6.right = std_mult_pipe_2.out;
      bb0_19[done] = ext_mem2_done;
    }
    group assign_while_0_latch {
      while_0_arg0_reg.in = std_add_2.out;
      while_0_arg0_reg.write_en = 1'd1;
      std_add_2.left = while_0_arg0_reg.out;
      std_add_2.right = 32'd1;
      assign_while_0_latch[done] = while_0_arg0_reg.done;
    }
    group assign_while_1_latch {
      while_1_arg0_reg.in = std_add_1.out;
      while_1_arg0_reg.write_en = 1'd1;
      std_add_1.left = while_1_arg0_reg.out;
      std_add_1.right = 32'd1;
      assign_while_1_latch[done] = while_1_arg0_reg.done;
    }
    group assign_while_2_latch {
      while_2_arg0_reg.in = std_add_0.out;
      while_2_arg0_reg.write_en = 1'd1;
      std_add_0.left = while_2_arg0_reg.out;
      std_add_0.right = 32'd1;
      assign_while_2_latch[done] = while_2_arg0_reg.done;
    }
  }
  control {
    seq {
      par {
        assign_while_2_init_0;
      }
      while std_slt_0.out with bb0_0 {
        seq {
          par {
            assign_while_1_init_0;
          }
          while std_slt_1.out with bb0_2 {
            seq {
              par {
                assign_while_0_init_0;
              }
              while std_slt_2.out with bb0_4 {
                seq {
                  seq {
                    bb0_6;
                    bb0_9;
                    bb0_12;
                    bb0_13;
                    bb0_15;
                    bb0_17;
                    bb0_19;
                  }
                  assign_while_0_latch;
                }
              }
              assign_while_1_latch;
            }
          }
          assign_while_2_latch;
        }
      }
    }
  }
}
```
Then I've used this command to try the RTL synthesis with Xilinx Vivado: `fud e --to resource-estimate matmul.futil`. From here the error mentioned above shows up, for other programs as well. But if I use `synth-files` it works perfectly fine.
I've looked up the `impl1` folder, there is no `main_utilization_placed.rpt` but a `matmul_utilization_placed.rpt`. So I changed the top module name to `main` and then the command works. So it can only work as https://docs.calyxir.org/lang/attributes.html?highlight=toplevel#toplevel introduced?

HahaLan97, 2023-02-23:
I think it would be nice to add support to more flexible naming mechanism rather than only one legal name `main`, which could cost additional effort.

rachitnigam, 2023-02-24:
Thanks for the bug report! We can do a couple of things to help with this for now
1. Short-term: Issue an error or warning when it is detected that we ran the synthesis flow with a non-main toplevel component
2. Medium-term: Wrap all designs without a `main` with a main component and make it the top-level
3. Long-term: Actually support different top-level designs.

The reason I mention (2) in addition to the more obviously correct (3) is that I'm not sure how hard it would be change the synthesis scripts to use a different "top" module.

sampsyo, 2023-02-25:
Right; I think the only weirdness about option 3 is that we would need to somehow customize the thing that's instantiated in the RTL emulation testbench on this line:
https://github.com/cucapra/calyx/blob/5fcf9efff5ae69239b688b0a72a993262a03951b/fud/icarus/tb.sv#L5

…or else just rename the module to `main` during codegen.

closed 2023-05-21 by rachitnigam