calyxir/calyx #1397: To-Do Tasks for Static Groups

calebmkim, 2023-03-11:
A tracker for the "to-do" tasks for Static Groups. Some are definitely bigger than others, and we should probably add issues as they come up. Right now, this is just a preliminary list. @paili0628 feel free to add more in case I forgot any. 
Note that much of the existing code will break once we turn the assignments field of static groups from `assignments` type -> `static_assignment` type, which is something we are keeping in mind. 

## Shorter Term (to do once we get the IR fully formed) 
- [ ] Passes that visit `Enables` might want to visit `StatcEnables`. This is a case by case basis, but some of the passes that apply operations to groups should apply them to static groups as well 
- [ ] There is some code duplication in LiveRangeAnalysis, Well Formed, and Reaching Definitions. Once we get the IR fully figured out, we should figure out the best way to reduce this duplicate code. 
- [ ] Add a latency field for static groups 

## Longer Term (to do once we implement static groups correctly) 
- [ ] Python Library (and other frontends) should be rewritten 
- [ ] We should change `infer-static-timing` to change groups -> static groups, instead of adding an attribute

paili0628, 2023-04-11:
I have another task but I'm not sure if it's short term or long term? The `compile-sync` pass does not take in `static` control now. We need to figure out what type of synchronization we want for `static` control.

sampsyo, 2023-04-12:
Ah, that makes sense!! Cool!!!

rachitnigam, 2023-05-11:
> group2seq is weird w/ static groups. How do we infer the latency of the split groups? We can leave it to infer-static-timing, but then, would that mean turning the static groups into 2 non-static groups?...

We've decided to not split apart static groups since its just the inverse transform of static inline

rachitnigam, 2023-05-11:
Once we're done with the Static groups milestone, we'll go over this again and create issues from all the relevant bullet points and close this tracker

rachitnigam, 2023-06-27:
@calebmkim @paili0628 can y'all go over this and update it?