calyxir/calyx #1405: Static Groups Syntax

calebmkim, 2023-04-07:
We have a couple choices for representing the latency of static groups in the syntax. 

1.  Use the done condition 
```
static group g {
    ... 
    g[done] = %4 
}
```
Here, the static group's latency is 4. One advantage is that we can directly see how the done conditions of static groups get translated to dynamic groups (i.e., we can clearly which FSM state will correlate to the done condition of the dynamic group). However, this syntax can also be a bit misleading, since in reality ` g[done] = %N` is the _only_  possible format for an assignment to a static group's done condition. For example, it can never be something like `g[done] = cell.done` since `g` is a static group. 

2. Use some sort of annotation  
```
static group g <4>{
    ... 
   // no assignment to g[done] 
}
```
This has the advantage that there is no generality of the done condition. The done condition of a static group can't be `cell.done` in this syntax. 

I've chosen 2) tentatively but am open to feeback.

sampsyo, 2023-04-10:
Good question! Since it just affects the concrete syntax, my main thought here is that it doesn't matter too much—both of these options would be totally fine and let's not worry too much about getting it wrong.

With that out of the way, we can bikeshed to our hearts' content! Without much of a reason at all, my gut reaction is to prefer something close to option 2, i.e., what you've already implemented. It seems nice to be able to see the latency of the group as part of the group's declaration. And `g[done] = %4` does seem *especially* fictional, whereas `g[done] = foo.out` in a dynamically-timed group is only *kinda* fictional (i.e., there actually is going to be a signal wired up there).

rachitnigam, 2023-04-11:
I agree that the whole `g[done] = %4` syntax is quite fictional even though it *looks nice*. Maybe we can use the other syntax for now with the understanding latency is a group's property and not due to some assignment. If we find a different syntax in the future that we like, we can switch to it!

rachitnigam, 2023-05-05:
How about something like:
```
static<4> group foo { ... }
```
So that we don't conflict with attributes for the group.

sampsyo, 2023-05-07:
I was kinda thinking about that too! It makes it a bit clearer what the integer is for.