calyxir/calyx #1406: Compiling Static Groups

calebmkim, 2023-04-07:
To make sure we're on the same page... ideally, there will be two stages to compiling static control: 

1. "Static Inlining" 

This takes static control statements (e.g., static seqs, pars, whiles, etc.) and turns them into, a single static group. Ideally, we want to inline such that eventually, each static group is "surrounded" by a dynamic context. Otherwise--if it's surrounded by a static context--that means we could do more inlining. 

Example: 
```
wires {
    static group one<1> { r.in = 10; r.write_en = %0 ? 1'd1}
    static group two<1> { a.in = 20; a.write_en = %0 ? 1'd1}
}
control { 
    static seq { one; two }
} 
```
into 
```
wires {
    static group compiled <2>{
          r.in = 10;
          r.write_en = %0 ? 1'd1;
          a.in = 20;
          a.write_en = %1 ? 1'd1;
    }
}
control { compiled }
```

2.   "Static to Dynamic" 
This takes the single, static group, and turns it into a dynamic group by instantiating an FSM for the static group. This is what #1403 implements. Example (from step 1): 
```
wires {
    group compiled {
          r.in = 10;
          r.write_en = fsm.out == 0 ? 1'd1;
          a.in = 20;
          a.write_en = fsm.out == 1 ? 1'd1;
          compiled[done] = fsm.out == 2 ? 1'd1; 
          fsm.in = fsm.out < 2 ? fsm.out + 1
    }
    fsm.in = fsm.out == 2 ? 0; 
}
control { compiled }
```

### A Different Way To Compile

One question is, what if we have a static seq with two static groups, but we want to compile static groups using two separate FSMs? We don't want to just compile them immediately into dynamic groups (i.e., skip step 1 and go straight to step 2). If we do that, then we would "waste" cycles reading from the done condition of each group. I'm thinking of something where we have separate FSMs for separate static groups, while maintaining static timing guarantees, which means no reading from done conditions. Example: 
```
wires {
    static group one<1> { r.in = 10; r.write_en = %0 ? 1'd1}
    static group two<1> { a.in = 20; a.write_en = %0 ? 1'd1}
}
control { 
    static seq { one; two }
} 
```
into 
```
wires {
    static group one<1> { r.in = 10; r.write_en = %0 ? 1'd1}
    static group two<1> { a.in = 20; a.write_en = %0 ? 1'd1}
    static group compiled <2>{
          one[go] = % 0 ? 1'd1; 
           two[go] = % 1 ? 1'd1; 
    }
}
control { compiled; } 
```
I'm not saying this is correct, but somethign like this may be able to accomplish what I was thinking...

sampsyo, 2023-04-10:
This is a really interesting thought; thanks for writing it up, @calebmkim!

To make the design space even more complete, I just want to mention that there is *another* style of hardware you can get by using a dynamic control operator instead of a static one. That is, if you were to do `seq { one; two }` instead of `static seq { one; two }`, you can imagine what hardware you'd get: it might look something like your final example but `compiled` would be dynamically-timed. Fortunately, there is an obvious syntactic way that Calyx programs can "request" this style of output hardware from the compiler.

To crystallize the question in this thread: how (if at all) should Calyx programs "request" the resulting hardware you show at the end of the post? That is, do Calyx programs ever want to avoid inlining but still do static control? It is not yet clear to me when this would be advantageous, but I agree that you can imagine that we will find such a reason in the future.

One possible strategy we can take for now: let's keep this issue as a reminder that we may want to support this "alternate" style of compilation in the future, but forge ahead with the first route for now? Maybe? Unless there is a good simplicity-related argument for taking the latter approach first!

rachitnigam, 2023-04-11:
A particular outcome of this process should be that we handle `seq` and `static seq` compilation separately. For example, the goal of the `static` inlining pass is to eliminate all `static` control operators in the program while the `compile-static` pass eliminates all the static groups from the program. 

Instead of the static inlining, we can imagine another way to eliminate control operators is manually allocating an FSM for each static group in such a way that the FSM resets in the last state and doesn't expose the done condition of the group. This corresponds to a "naive" compilation strategy where we have two way to compile static groups: when they are in a "dynamic context" (`seq`, `par`, etc.) allocating an FSM which can provide a `done` condition for the generated dynamic group and when in the "static context" (`static seq`, etc.) generating an FSM which is time-sensitive and does not expose a done condition for the group.

calebmkim, 2023-04-14:
I was just thinking about compiling while loops, and I think this discussion is relevant, so I'm making a note about it here. 

If we have something like: 

```
static repeat n {
  static_body; // let's say latency = 4  
}
```

In order to not "waste" a cycle after `static_body`, we don't want `static_body`'s fsm to ever reach its latency (in this case, 4). That is, we want it to reset to 0 as soon as it reaches 3. 

However, in the `compile-static` pass that I pushed, we don't assignments to do the "early fsm reset" as part of the assignments in the group, since then it would never reach its done condition. This would be problematic since the static group may appear elsewhere in the control, e.g., in a dynamic context in which its done condition may be needed. 

We only want to do the "early resets" depending on the context in which the `static enable` appears. Therefore, I was thinking that we could have an attribute `early_reset` that the compiler can attach to control statements and internally use for helping compilation. It could work something like this:  

```
static repeat n {
  // @early_reset attribute tells us that we should reset static_body's fsm early 
  @early_reset() static_body; // let's say latency = 4  
}
```

Then, we will compile this repeat/static while loop doing the following: 

```
static_repeat_group <n * 4>{
  // "big fsm" that counts up to n * 4. Note that this is different than static_body_fsm 
  static_body_fsm.in = static_body_fsm.out == 3 ? 0 // resets static_body early 
  // inline all of static_body's assignments. We should inline after we have already 
  // defined static_body's fsm, though. 
}
```

In fact, I think this may be able to work for unbounded/dynamic while loops with static bodies. I'm not 100% sure this is correct, but I can see something like 

```
while port {
  @early_reset() static_body; // let's say latency is 4 
}
```

compiling to 

```
group while_group {
  // inline all of static_body's assignments. To do this, though, the assignments 
  // need to be compiled to dynamic first (i.e., turn %3 ? into fsm.out == 3 ?)
  fsm.in = fsm.out == 3 ? 0 // reseting the fsm   
  while_group[done] = fsm.out == 0 && !port ? 1'd1 
}
```

To determine whether this attribute-based way of thinking will ultimately work, I think I probably have to just try it out and see if I can make it work, although I am open to ideas and suggestions. An important note, though, is that any enable with an `early_reset` attribute can't appear in any context where we would need the done condition (in a dynamic seq, for example). It could appear in a `static_seq` though, and potentially help us with this "separate fsm compilation" stuff mentioned here: e.g., `static seq { @early_reset  g1; @early_reset g2; @eartly_reset g1 }`

sampsyo, 2023-04-16:
Hmm; interesting! I confess it raises a bit of a "yellow flag" in my mind that the style of compilation for a certain group depends on the context where it is used. I understand it might be necessary, but I think it would be good to convince ourselves that it really is absolutely necessary and there isn't a clever way around it (i.e., some "universal" way to compile groups that works both when they are the only (last?) group in a `repeat` body and otherwise). In other words, there is a significant cost in complexity here, so let's make sure the benefit is worth the cost.

Returning to this:

> However, in the `compile-static` pass that I pushed, we don't assignments to do the "early fsm reset" as part of the assignments in the group, since then it would never reach its done condition. [...]

To expand on why I think this dangerous: the problem is that we're hoping to compile groups like `static_body` so that _they themselves_ contain something roughly like `fsm.in = fsm.out == 3 ? 0`. This of course only makes sense `static_body` is in a loop; otherwise, we want to transition to a different state (4) where we emit a done signal for the group.

One thing about the above observation, however, is that the done signal only actually matters for the "static in dynamic" case. That is, we only need the done signal when wrapping up a static island for use in a dynamic ocean. Could we decouple this done-signal-adding so it only happens in that circumstance? So in other circumstances, we can always do the "don't waste a cycle" thing?

I don't think I described this very well, and it's possible that a synchronous discussion would be better!

calebmkim, 2023-04-17:
Yeah, I definitely understand what you're saying. 

I guess another way of framing the challenge is this: in the `static inlining` pass, we don't instantiate the FSMs yet; all we are working with is the `%` syntax. So there's no way to write the `fsm.in = fsm.out == 3 ? 0` assignment. However, when we _eventually_ instantiate the FSMs in the future, we want to be able to do the "early reset"/`fsm.in = fsm.out == 3 ? 0` stuff in some cases, but not in others. 

I proposed the attribute as a way of figuring out when to do this. But it seems like there's probably a smarter way to do this: we pretty much always want to do the "early reset" stuff _except_ in the "static in dynamic" case. Therefore, we can probably just figure out if we want to do the "early reset" stuff based on analyzing the control flow at the time we instantiate the FSMs, without any need for the attribute. Thinking this over, I think you're probably right; the attribute stuff is probably not the best way to do this.

calebmkim, 2023-04-17:
Ok, so based on an idea from @sampsyo that we discussed synchronously, we decided (for now) on the following idea: 

We can first do static inlining as much as we want. Then we compile static groups to _always_ "reset early": this means their done condition will never be reached.  However, some of these static groups may appear in a dynamic context, in which case we _want_ it to reach its done condition. WE do an analysis to determine for which groups this applise to, then we add a "wrapper" dynamic group. Something like: 

``` 
group  g { 
    // a bunch of assignments 
    fsm.in == fsm.out == 3 ? 0
    g[done] = fsm.out == 4 ? 1'd1
}
```
into 
```
group  g_wrapper { 
    g[go] = 1'd1 ;
    g_wrapper[done] = // something that denotes "the second time g's fsm reaches 0 (i.e., the first time it resets)" 
}
```

PR #1418 (2023-04-27): Full Static Control (Except Invoke) Inlining & Compilation 
Adds static `seq`, `par`, `if`, and `repeat` support, in addition to static `enable`!  

Currently, it inlines each of these four different control operators into a single static group in the `static-inline` pass, and then the `compile-static` pass 1) instantiates FSMs so that these inlined static groups get turned into dynamic groups with "early-resetting" FSMs that never reach the group's done condition, and 2) creates wrapper groups for these "early reseting" groups when necessary (i.e., when they need to use a go-done interface).  [... cut]
closed 2023-04-27 by calebmkim