calyxir/calyx #1407: comb-prop Pass wrongly optimizes away wires

matth2k, 2023-04-10:
I have `std_wire` cells being wrongly optimized away. These wires are both driven and used without a group. Once I disable the `comb-prop` pass, I can see the missing signals in the verilog and my verilator simulation works as expected. Otherwise, the handshake signal is tied to 0, and my simulation hangs.

In the below example, one such signal being optimized away is `merge_hw_1_write_done_port2 = std_wire(1);`

I could try to find a minimal test case with the issue if that's needed.

```
import "primitives/core.futil";
import "primitives/binary_operators.futil";
extern "ram_1r1w.sv" {
  primitive ram_1r1w[WIDTH, PORT0_SIZE, PORT0_IDX_SIZE, PORT1_SIZE, PORT1_IDX_SIZE](@clk clk: 1, port0_addr0: PORT0_IDX_SIZE, @go @static(1) port0_read_en: 1, port1_addr0: PORT1_IDX_SIZE, port1_write_data: WIDTH, @go(2) @static(1) port1_write_en: 1) -> (@stable port0_read_data: WIDTH, @done port0_read_done: 1, @done(2) port1_write_done: 1);
}
comb component merge_hw_0(merged_addr0: 3) -> (shared_addr0: 2, port1_select: 1, port2_select: 1) {
  cells {
    port2_select_eq = std_eq(1);
    port1_select_eq = std_eq(1);
    shared_addr_slice = std_slice(3, 2);
    select_bits_slice = std_slice(3, 1);
    select_bits = std_rsh(3);
  }
  wires {
    port2_select_eq.left = select_bits_slice.out;
    port2_select_eq.right = 1'd1;
    port2_select = port2_select_eq.out;
    port1_select_eq.left = select_bits_slice.out;
    port1_select_eq.right = 1'd0;
    port1_select = port1_select_eq.out;
    shared_addr_slice.in = merged_addr0;
    shared_addr0 = shared_addr_slice.out;
    select_bits_slice.in = select_bits.out;
    select_bits.right = 3'd2;
    select_bits.left = merged_addr0;
  }
}
comb component merge_hw_1(merged_addr0: 3) -> (shared_addr0: 2, port1_select: 1, port2_select: 1) {
  cells {
    port2_select_eq = std_eq(1);
    port1_select_eq = std_eq(1);
    shared_addr_slice = std_slice(3, 2);
    select_bits_slice = std_slice(3, 1);
    select_bits = std_rsh(3);
  }
  wires {
    port2_select_eq.left = select_bits_slice.out;
    port2_select_eq.right = 1'd1;
    port2_select = port2_select_eq.out;
    port1_select_eq.left = select_bits_slice.out;
    port1_select_eq.right = 1'd0;
    port1_select = port1_select_eq.out;
    shared_addr_slice.in = merged_addr0;
    shared_addr0 = shared_addr_slice.out;
    select_bits_slice.in = select_bits.out;
    select_bits.right = 3'd2;
    select_bits.left = merged_addr0;
  }
}
component mem_0_comp(@clk clk: 1, port0_addr0: 3, @go @static(1) port0_read_en: 1, port1_addr0: 3, port1_write_data: 32, @go @static(1) port1_write_en: 1, @reset reset: 1) -> (@stable port0_read_data: 32, @done port0_read_done: 1, @done port1_write_done: 1) {
  cells {
    merge_hw_1_port_2_write_en_gated = std_and(1);
    merge_hw_1_port_1_write_en_gated = std_and(1);
    merge_hw_1_write_dones_or_d1_i0 = std_or(1);
    merge_hw_1_write_done_port1 = std_wire(1);
    merge_hw_1_write_done_port2 = std_wire(1);
    merge_hw_inst1 = merge_hw_1();
    merge_hw_0_port_2_read_en_gated = std_and(1);
    merge_hw_0_port_1_read_en_gated = std_and(1);
    merge_hw_0_data_latch_predicate_false = std_not(1);
    merge_hw_0_data_latch_new_value = std_wire(32);
    merge_hw_0_data_latch_module_value = std_wire(32);
    merge_hw_0_data_latch_predicate_true_d1_i0 = std_or(1);
[... 180 lines cut ...]
        seq {
          bb0_3;
          assign_while_0_latch;
        }
      }
      assign_while_1_init_0;
      while std_slt_1.out with bb0_4 {
        seq {
          bb0_6;
          bb0_8;
          assign_while_1_latch;
        }
      }
      assign_while_2_init_0;
      while std_slt_2.out with bb0_9 {
        seq {
          bb0_11;
          bb0_12;
          assign_while_2_latch;
        }
      }
    }
  }
}
```

matth2k, 2023-04-11:
> Have you tried minimizing the program a bit to see what causes the problem?

Tried just now for a little bit. A mv program compiles to 2 groups, but the issue is no longer present. Same with mv in a loop. I also think that I'm now hitting a bug on `inter-static-timing`. 

Some programs I get 
```
Error: Malformed Structure: Impossible "static" latency annotation for component mem_0_comp.
Annotated latency: 1
Inferred latency: 0
```
even with `-d comb-prop`.

For now, I'm running with `-d inter-static-timing` to find a minimal test case.

matth2k, 2023-04-11:
`component mem_0_comp(@clk clk: 1, port0_addr0: 3, @go @static(1) port0_read_en: 1, port1_addr0: 3, port1_write_data: 32, @go @static(1) port1_write_en: 1, @reset reset: 1) -> (@stable port0_read_data: 32, @done port0_read_done: 1, @done port1_write_done: 1) `

Is this component a problem since the go/done pairs are not labeled with an index?

rachitnigam, 2023-04-11:
The pretty printer  does a silly thing where `@go` and `@go(1)` are the same.

matth2k, 2023-04-11:
> The pretty printer does a silly thing where `@go` and `@go(1)` are the same.

I still think it's a mistake on my part, because there are two interfaces there. So it should be `@go` and `@go(2)`. However, with that fix I'm not seeing any difference behavior.

At the least, I've realized that the compute component has nothing to do with the bug. Meaning that if you chop off the main component, the bug still persists. 

Finally, the wires being deleted have no guarded assignments and are not used in groups. So it makes sense that the compiler is trying to shortcircuit/bypass them. For some reason, only the write interface is having the problem

```bash
$ cat vadd_merge_fix.futil | grep i0.left
    merge_hw_1_write_dones_or_d1_i0.left = merge_hw_1_write_done_port1.out;
    merge_hw_0_data_latch_predicate_true_d1_i0.left = merge_hw_0_read_done_port1.out;
    merge_hw_0_read_dones_or_d1_i0.left = merge_hw_0_read_done_port1.out;
$ ./target/debug/futil  -p well-formed -p post-opt vadd_merge_fix.futil | grep i0.left
    merge_hw_0_read_dones_or_d1_i0.left = mem_0_comp_bank_1.port0_read_done;
```

rachitnigam, 2023-04-11:
Interesting! Couple of things:
- Can you add the minimal version of the code you're working with that have the read bypass problem?
- Can you open a separate issue for the example that runs into the "impossible latency" error? That's the compiler telling you that a static group has been given an incorrect latency annotation so either the CIRCT compiler is wrong or we are making a mistake.
- You can `-x comb-prop:no-eliminate` to see which assignments the pass is trying to delete. In a nutshell, it tries to inline wires that have only one writer and then delete the assignments that are dead. 

Also, pro-top, all of the special options in the compiler are documented if you run `cargo run -- --list-passes`. The `no-eliminate` option is explained there.

matth2k, 2023-04-11:
> * Can you add the minimal version of the code you're working with that have the read bypass problem?

This is as simple as I could get it:

```
import "primitives/core.futil";
extern "ram_1w.sv" {
  primitive ram_1w[WIDTH, SIZE, IDX_SIZE](addr0: IDX_SIZE, @clk clk: 1, write_data: WIDTH, @go @static(1) write_en: 1) -> (@done write_done: 1);
}
comb component merge_hw_0(merged_addr0: 3) -> (shared_addr0: 2, port1_select: 1, port2_select: 1) {
  cells {
    port2_select_eq = std_eq(1);
    port1_select_eq = std_eq(1);
    shared_addr_slice = std_slice(3, 2);
    select_bits_slice = std_slice(3, 1);
    select_bits = std_rsh(3);
  }
  wires {
    port2_select_eq.left = select_bits_slice.out;
    port2_select_eq.right = 1'd1;
    port2_select = port2_select_eq.out;
    port1_select_eq.left = select_bits_slice.out;
    port1_select_eq.right = 1'd0;
    port1_select = port1_select_eq.out;
    shared_addr_slice.in = merged_addr0;
    shared_addr0 = shared_addr_slice.out;
    select_bits_slice.in = select_bits.out;
    select_bits.right = 3'd2;
    select_bits.left = merged_addr0;
  }
}
component mem_0_comp<"toplevel"=1>(@clk clk: 1, port0_addr0: 3, port0_write_data: 32, @go @static(1) port0_write_en: 1, @reset reset: 1) -> (@done port0_write_done: 1) {
  cells {
    merge_hw_0_port_2_write_en_gated = std_and(1);
    merge_hw_0_port_1_write_en_gated = std_and(1);
    merge_hw_0_write_dones_or_d1_i0 = std_or(1);
    merge_hw_0_write_done_port1 = std_wire(1);
    merge_hw_0_write_done_port2 = std_wire(1);
    merge_hw_inst0 = merge_hw_0();
    mem_0_comp_bank_0 = ram_1w(32, 4, 2);
    mem_0_comp_bank_1 = ram_1w(32, 4, 2);
  }
  wires {
    mem_0_comp_bank_0.clk = clk;
    mem_0_comp_bank_1.clk = clk;
    merge_hw_0_port_2_write_en_gated.left = merge_hw_inst0.port2_select;
    merge_hw_0_port_2_write_en_gated.right = port0_write_en;
    merge_hw_0_port_1_write_en_gated.left = merge_hw_inst0.port1_select;
    merge_hw_0_port_1_write_en_gated.right = port0_write_en;
    port0_write_done = merge_hw_0_write_dones_or_d1_i0.out;
    merge_hw_0_write_dones_or_d1_i0.left = merge_hw_0_write_done_port1.out;
    merge_hw_0_write_dones_or_d1_i0.right = merge_hw_0_write_done_port2.out;
    merge_hw_inst0.merged_addr0 = port0_addr0;
    mem_0_comp_bank_0.write_data = port0_write_data;
    mem_0_comp_bank_0.write_en = merge_hw_0_port_2_write_en_gated.out;
    merge_hw_0_write_done_port2.in = mem_0_comp_bank_0.write_done;
    mem_0_comp_bank_0.addr0 = merge_hw_inst0.shared_addr0;
    mem_0_comp_bank_1.write_data = port0_write_data;
    mem_0_comp_bank_1.write_en = merge_hw_0_port_1_write_en_gated.out;
    merge_hw_0_write_done_port1.in = mem_0_comp_bank_1.write_done;
    mem_0_comp_bank_1.addr0 = merge_hw_inst0.shared_addr0;
  }
  control {
  }
}
```

> * That's the compiler telling you that a static group has been given an incorrect latency annotation so either the CIRCT compiler is wrong or we are making a mistake.

This same code also shows the Inferred latency: 0 problem. So it isn't CIRCT being a problem; it would be AMC's fault. Basically the `mem_0_comp` does some combination stuff on 2 memory primitives with latency 1. So the end latency of the top-level go/done pairs should still be 1. 

> * You can `-x comb-prop:no-eliminate` to see which assignments the pass is trying to delete.

```bash
$ ./target/debug/futil  -p well-formed -p post-opt -x comb-prop:no-eliminate reduced.futil | grep dead
    @dead merge_hw_0_write_dones_or_d1_i0.left = mem_0_comp_bank_1.write_done;
    @dead merge_hw_0_write_dones_or_d1_i0.right = mem_0_comp_bank_0.write_done;
    @dead merge_hw_0_write_dones_or_d1_i0.right = mem_0_comp_bank_0.write_done;
    @dead merge_hw_0_write_dones_or_d1_i0.left = mem_0_comp_bank_1.write_done;
```

It think's the external primitive's done signal is constant 0?

rachitnigam, 2023-04-19:
Okay, so the problem is that `comb-prop` is yet again a bit overzealous in its reasoning. The problem is that when it observes something like this:
```
w.in = x;
```

It thinks that all uses of `w.out` need to be replaced with `x` and assumes it can mark the assignment for removal. When it sees something like this:
```
y = w.out;
```

It thinks that all uses of `w.in` need to be replaced with `y` and marks the assignment for removal. Sadly, when you have something like:
```
w.in = x;
y = w.out;
```

It happily transforms it into:
```
@dead y = x;
@dead y = x;
```

Which is problematic because now the assignment has disappeared altogether.

closed 2023-04-27 by rachitnigam