calyxir/calyx #1408: Getting incorrect result running Calyx tutorial example

gabizon103, 2023-04-12:
I followed the [Getting Started](https://docs.calyxir.org/#getting-started) tutorial and was able to successfully install everything needed - running `fud check` tells me that everything besides `interpreter`, `dahlia`, `verilog`, `vcd`, `synth-verilog`, and `vivado-hls` was installed correctly. 

However, when I run the given command
 `fud e examples/tutorial/language-tutorial-iterate.futil \
  -s verilog.data examples/tutorial/data.json \
  --to dat --through icarus-verilog -v` , I do not get the expected result. 

I get:
`{
  "cycles": 100,
  "memories": {
    "mem": [
      10
    ]
  }
}`

The expected result is:
`{
  "cycles": 100,
  "memories": {
    "mem": [
      42
    ]
  }
}`

Clearly, the value in memory is not being updated, but I'm not sure why. If anyone has seen this before or has ideas about what might be going wrong, I would really appreciate some help! (and apologies in advance if this is a silly fix, which I feel it might be).

EclecticGriffin, 2023-04-12:
Seems to be an issue with icarus in this case, the `verilog` and `interpreter-out` targets produce the expected results. If I had to guess without looking at a waveform I'd suspect it was an x-simulation thing, that's often the culprit in these cases.

rachitnigam, 2023-04-12:
Argh, that's weird. Can you try using `verilator` instead by passing `--through verilog`?

gabizon103, 2023-04-13:
That didn't work either, but for a different reason. Here's the output I got:

[image]

I wasn't sure if this was due to an issue with my installation, but I reinstalled verilator and got this same issue.

rachitnigam, 2023-04-13:
Can you run the “fud” command with the extra option “-vv”? That’ll give you some more information

gabizon103, 2023-04-13:
Here's what I got: 

[image]

gabizon103, 2023-04-14:
[Updating to gcc-10](https://veripool.org/guide/latest/exe_verilator.html#cmdoption-timing) fixed this. Still not sure why icarus isn't working, but at least I can run Calyx programs now. Thanks!

rachitnigam, 2023-04-14:
Oh, good find! What was the previous version you were using? I think a lot of us have been using clang but kinda surprising that old gcc versions don’t work

rachitnigam, 2023-04-19:
Okay, in that case we should probably have some sort of `fud check` warning when the GCC version is less than 10.0?