calyxir/calyx #1413: Documentation for Python eDSL

rachitnigam, 2023-04-20:
We should add a short tutorial and reference documentation for the Python eDSL for Calyx

anshumanmohan, 2023-05-23:
Priya and I chatted at some length about this today. There are a few options for a "working example":
- [`builder_example.py`](https://github.com/cucapra/calyx/blob/master/calyx-py/test/builder_example.py)
- [`red_tree.py`](https://github.com/cucapra/calyx/blob/master/calyx-py/test/red_tree.py)
- [`arb.py`](https://github.com/cucapra/calyx/blob/master/calyx-py/test/arb6.py)
- MrXL's [`gen_calyx.py`](https://github.com/cucapra/calyx/blob/master/frontends/mrxl/mrxl/gen_calyx.py)

On one end, I think `builder_example` is a little skinny. Folks will "get it" 100%, which is nice, but when it comes to the competition they'll immediately run into roadblocks and they'll need us to wander around and whisper magic incantations that they could not have possible figured out. A little like how I needed Griffin and Adrian.

On the other end, the [frontend tutorial](https://docs.calyxir.org/tutorial/frontend-tut.html) does sketch how `gen_calyx` works, but it's involved enough that I see it as a glossary when working on MrXL extensions, not something to be understood end to end. 

My proposal is:
1. Carefully work through `red_tree` and `arb` and make a minimal list of all the builder tricks that were needed.
2. Bulk out the builder example carefully and minimally to show how each such trick works. Keep it to one trick per line, if possible.
3. Then the handout can literally be a landscape piece of paper, split down the middle, with a .py code block on the left and a .futil code block on the right. We can draw lines between them to show which lines of Python generated which lines of Calyx.

priyasrikumar, 2023-05-23:
Yes! Anshuman and I had a list of things that would be good to demonstrate in the builder tutorial, listed below: 

- saving handles to components/groups for later reference
- multiple components 
- invocations of other components 
- external mems: how to build them + the need for `.data` files 
- how to read/write/reference ports from registers 
- how to read/write memory 
- how to instantiate inputs/outputs of a component
- continuous assignments 
- guarded assignments
- possibly working with ref memories? 

I'm working on a new example that is hopefully a little less involved than the ones we're using for the MrXL tutorial-- in essence, we'll have one component that computes the [tetrahedral number](https://oeis.org/A000292) of a given number, and a main component that populates an external memory with the tetrahedral numbers of a given range of numbers. I think I can get the vast majority of items on the above list covered with this example. 

I like the idea of having a piece of paper with actual lines between the Python and generated Calyx, but I think the program might be a bit too long-- I'll try to style the documentation so that it mimics that approach.

Let me know if y'all have thoughts on any of the above-- otherwise, stay tuned for that example soon!!

rachitnigam, 2023-05-23:
Sweet! I would say that we should wrap this up ASAP because we need time to dry run the whole tutorial and make sure we have enough time for a second iteration pass to close up all the gaps

anshumanmohan, 2023-05-23:
Also, a reminder to fix the links from the frontend tutorial to the DSL docs. They go every which way right now

anshumanmohan, 2023-05-23:
Just wondering, is the tetrahedral number still in your plans? In #1503 you have explained the builder_example as it stands, yeah? 

If you do flesh out the builder example to show off more tricks, my suggestion is to resist the temptation to have a well-motived "story". My suggestion is to do something immediately understandable, like "we're going to add two numbers, but in a somewhat contrived way".

priyasrikumar, 2023-05-23:
The code ended up being way longer than I'd hoped, and I'm running into a very fixable bug that i just need to look at more, so I'm not sure I'll get it in in time for the tutorial (but I'll keep poking at it!). For your reference, the tetrahedral code is [here](https://github.com/cucapra/calyx/blob/tetra-builder/calyx-py/test/builder_tutorial.py). 

I'll keep working on building up the builder example as well. Could you clarify where links to the eDSL in the frontend tutorial are broken? Thank you!!

anshumanmohan, 2023-05-23:
IMO it's fine to leave tetrahedral for now. I think we want something more complicated than `builder_example` but less complicated than `red_tree`, and this is already well past `red_tree`.

anshumanmohan, 2023-05-23:
Just take a look at https://docs.calyxir.org/tutorial/frontend-tut.html and make sure that the various links go where you want? There isn't a _broken_ link, but they just don't do anything very thoughtful right now

anshumanmohan, 2023-05-24:
Reopening: the PR that was merged did not complete these goals

anshumanmohan, 2023-05-24:
Feel free to change to a new milestone, but I didn't want the discussion here to get relegated to the archives.

anshumanmohan, 2023-05-24:
I think that would be the gold standard, but it would require much more care.

priyasrikumar, 2023-05-24:
I can definitely transform the doc into more of a language reference! I think most of the material is there, it's just a matter of changing the format. I can make a PR for that today.

anshumanmohan, 2023-05-24:
@priyasrikumar I think you'll need to do a little more than that. Further features need to be documented, plus you'll need short examples (which don't need to be runnable, just inlined). See here: https://docs.calyxir.org/lang/ref.html?highlight=language#calyx-language-reference

I think a good litmus test would be: after reading this document that you're creating and the calyx_builder example + doc, can a reader work through reduction_tree.py and arbiter_6.py and have 0 surprises?

priyasrikumar, 2023-05-24:
yep, we discussed exactly this just now in the meeting. we are on the same page!

anshumanmohan, 2023-05-24:
Many thanks. Just helps to have it written down, with so many things flying around!

anshumanmohan, 2023-06-05:
Closed by #1525

closed 2023-06-05 by anshumanmohan