calyxir/calyx #1417: Static Invokes, Components, Primitives

calebmkim, 2023-04-24:
This may be incomplete, so feel free to add on @paili0628 ... 

We want to add static `invoke`s but this requires the notion of static `components`: we should only be able to statically `invoke` components that are themselves considered "static". Ideally, we would do this by analyzing the control flow, and being able to infer the static latency of the component's entire control flow, sort of like `infer-static-timing` does now. 

Furthermore, we want the notions of static primitives (e.g., we know registers take one cycle to write to). 
One question is what we should do with components with multiple go-done interface, such as memories. What if they have different latencies? We should attach an attribute to their ports, although we should probably call it something other than "static" to avoid confusion with static control. 

Even more abstractly (and even more in the future) we talked about about the potential for "virtual" components: frontends target Calyx using "virtual" hardware, such as a "virtual" multiplier. Then the Calyx compiler determines, based on the design/other information, what actual multiplier primitive would be best. This could also be applied to memory and would affect memory latencies.

rachitnigam, 2023-05-08:
Couple more thoughts from thinking about this with @paili0628. 

## Static Components

A static component has a defined static latency and is only allowed to use static groups and control. Furthermore, the only way to `invoke` such a component is through a `static invoke`. It is a well-formedness error to directly `invoke` it from a non-static context.

Much like static groups, we might need a "wrapper" to correct use such components from non-static contexts. Similarly, another guarantee of a static component should be that it's done condition is not visible and the allocated FSM has the "early resetting" behavior we desire.

## Static Primitives

A static primitive is similar to a static component in the sense that the only way to `invoke` it or use it is through `static invoke` and `static group` operations. It should *probably* be an error to use a static primitive from a non-static group but I'm not certain about this.

A big, open question here is how to represent primitives that have both a static and a dynamic interface, for example registers. Registers have a `go`-`done` interface but also define a latency. In this case, what's the right solution? Should we make registers fully `static` and generate wrappers every time they are used in a dynamic context? Or should we come up with some other representation?

sampsyo, 2023-05-08:
This is a good question:

> It should _probably_ be an error to use a static primitive from a non-static group but I'm not certain about this.

To spell out the reasons why this might be a good or bad idea to allow (for primitives and non-primitive components alike, I think):

* On one hand, it is weird to use a static component in a non-static group because that group would need to use the component's "static interface." That is, the group would need to do its own FSM-based time-counting to know when the static component is done. (Because it can't use `%4` stuff to do the same work.)
* On the other hand, why not allow it? It is weird to want to do that, but you can imagine cases where a frontend wants to generate a blob of "RTL"-like code in a dynamic group that happens to use a static component in exactly that way. It doesn't seem to cause a _problem_, exactly, even if it is weird.

So I can honestly see this going either way!

And yeah, this is also very interesting:

> A big, open question here is how to represent primitives that have both a static and a dynamic interface, for example registers. Registers have a `go`-`done` interface but also define a latency. In [...]

Another solution, not that it's a good one, is that we just have two entirely different primitives: `std_reg` and `std_static_reg`, the latter of which does not have a `go`/`done` interface.

closed 2023-05-13 by calebmkim