calyxir/calyx #1429: Deprecate `static` attribute

rachitnigam, 2023-04-30:
Once the transition to static groups and control is completed, we should deprecate the existing `"static"` attribute completely and remove/change passes that make use of it. One big blocker that requires coordination with @andrewb1999 is changing the CIRCT frontend to not use the static attribute anymore.

calebmkim, 2024-01-16:
> Remove the `attr-promotion` pass (not sure what is does and why it needs to exist)

@paili0628 can correct me if I'm wrong, but I think `attr-promotion` takes groups with a `@static(n)` annotation--given to it by a frontend, and upgrades it to a `static<n>` group. Whereas `static-promotion` infers `static<n>` from a plain, non-annotated groups. Tbh I think it probably makes sense for `attr-promotion` to be merged with `static-promotion`, but it doesn't really matter now that we're gonna deprecate `@static`.

calebmkim, 2024-01-16:
One thing I was wondering: how should frontends generate Calyx, with the `@static` attribute now deprecated?  Obviously, if there is some very specific cycle level timing behavior you want, you should use `static`. 

But if there isn't, do we want to generate `static<n>` groups but dynamic `control`, so that compaction opportunities can still occur? Or just dynamic groups and control, and rely on `promotion` and `compaction`? Lmk if this question makes sense.

rachitnigam, 2024-01-16:
We talked about (but don't have an open issue) on another attribute called `@loose` or `@reschedulable` on `static` control programs to tell the compiler that it's okay to reschedule the things. 

The other option is exposing the `@promotable` attribute but need to think carefully about the interactions. Either way, we need some way to express "this block has a timing information available but doesn't rely on guarantees". It would be good if it is not a completely different and overlapping mechanism like `@static`.

calebmkim, 2024-01-16:
Sounds good, I can try to think about this some more and we can discuss in the Calyx meeting next Monday.

sampsyo, 2024-01-22:
Just as a little preparation for this upcoming synchronous meeting: @calebmkim's proposal along these lines makes sense to me:

> do we want to generate `static<n>` groups but dynamic `control`

It makes sense to me for frontends to do this (i.e., generate static "leaves" with a plain dynamic "tree"). Basically, it seems to me like static control inference works well enough that it's not clear why a frontend would ever need to generate a `static` control statement—with the very important exception of cases where the semantic timing guarantees matter. (That is, when interfacing with an external system.)

I could be wrong, but I think the excuse we would need to implement an externally-supported `@reschedulable` or whatever would be a concrete example of a case where the frontend needs to provide this hint, and automatic inference doesn't work well.

calebmkim, 2024-01-25:
We've had a few synchronous discussions since this post, so I'm going to try to summarize the basic takeaways/calls to action from them. 

### Separating Inference and Promotion
We want to separate `inference` and `promotion` into separate passes. I have a PR ready that will do that. One important problem we tackle: suppose component `B` instantiates component `A`. Suppose also that we infer component's `A`'s latency, and use that to infer the latency of `B`. If we decide not to promote `A`, then we can the inference information in `B` is no longer valid. This requires a "Fix Up". 

### Compaction goes in between inference and promotion. 
Once we have done that, we may want to insert `compaction` in between the `inference` and `promotion` passes. There are two challenges to keep in mind. 

1. Suppose again that component `B` instantiates component `A`. Suppose also that we infer component's `A`'s latency, and use that to infer the latency of `B`. If we decide to compact `A`, thereby decreasing its inferred latency, then the inference information in `B` is no longer valid. This requires a similar "Fix Up" to the original problem. 
2. Compaction requires two things: a) timing guarantees to ensure dependencies are obeyed in our schedule and b) the ability to reschedule groups to make it more compact. a) is a property of static control but not dynamic control, while b) is a property of dynamic control but not static control. The proposal we came out of in the meeting today was the following: `compaction` will take in dynamic control annotated with the `@promotable` attribute, and it will emit `static<n>` code. In other words, it does the same thing as promotion, but in a more specific context and manner. Compaction will be very opportunistic (to start, we will probably compact any `seq` whose latency we can decrease, even if it's only by one cycle). 

### Names of Attributes 
The proposal above suggests using `@latency(n)` as the attribute to attach to components' `go` ports. We were thinking that a slightly better name would be `@interval`. The reason is the following. Suppose we define a pipelined component in Calyx: the number of cycles its control flow will take is the II, not the latency, of the component. For non-pipelined components, latency and II will obviously be the same. But I think `@interval` is maybe a slightly better name for the attribute?

rachitnigam, 2024-01-26:
@calebmkim can we also summarize the discussion between the `@loose` annotation (applied to `static` programs) and `@promotable` (applied to dynamic programs)? I think that might be relevant to the "taxonomy" of static programs that we were discussing synchronously

calebmkim, 2024-01-26:
Sounds good, just claimed the next slot

rachitnigam, 2024-02-12:
@calebmkim can we update this w.r.t. #1897?

calebmkim, 2024-02-12:
Sure, I've checked the boxes I've updated.  Couple notes: 

- `@latency` renamed to `@interval` 
-  kept `compute-static` bc it is needed for `InferenceAnalysis` (i.e., repropogating `@promotable` throughout the control when necessary)

rachitnigam, 2024-02-12:
Awesome! Looks like this issue is completed then? Maybe we need to update the docs for mentions of `@static` and update it?

calebmkim, 2024-02-12:
Sure, I can update docs

PR #1911 (2024-02-12): Remove `@static` from docs
Closes  #1429. 

Lmk if you think any more detail is needed on these.