calyxir/calyx #1434: Non-synthesizable verilog should be guarded with preprocessor macros

matth2k, 2023-05-02:
When I use the verilog backend, `verilog.rs` generates some blocks to check for invalid assignments.

https://github.com/cucapra/calyx/blob/1b8bd1dfdd196f9df653a9160e896d60a0fe5746/src/backend/verilog.rs#L496-L508

However, Xilinx Vivado has a problem with this because its not synthesizable code.

I think the solution is to simply add some macros so Vivado skips over those lines like such:
```
`ifdef VERILATOR
  `ifndef SYNTHESIS
    // Non-synthesizable calls here
    always_comb begin
    if(~$onehot0({_guard465, _guard462})) begin
      $fatal(2, "Multiple assignment to port `while_2_arg0_reg.in'.");
    end
    end
  `endif
`endif
```

rachitnigam, 2023-05-02:
You have to pass `--disable-verify --synthesis --disable-init` to the compiler to remove these (see #1400)

rachitnigam, 2023-05-02:
Closing this in favor of #1400

closed 2023-05-02 by rachitnigam