calyxir/calyx #1435: AMC test confuses infer-static-timing pass

matth2k, 2023-05-02:
`infer-static-timing` errors on on one of my AMC tests:

```bash
$ futil -b verilog testIST.futil 
Error: Malformed Structure: Impossible "static" latency annotation for component mem_0_comp.
Annotated latency: 1
Inferred latency: 0
```

Here is the input I'm using:
```
import "primitives/core.futil";
extern "ram_1w.sv" {
  primitive ram_1w[WIDTH, SIZE, IDX_SIZE](addr0: IDX_SIZE, @clk clk: 1, write_data: WIDTH, @go @static(1) write_en: 1) -> (@done write_done: 1);
}
comb component merge_hw_0(merged_addr0: 3) -> (shared_addr0: 2, port1_select: 1, port2_select: 1) {
  cells {
    port2_select_eq = std_eq(1);
    port1_select_eq = std_eq(1);
    shared_addr_slice = std_slice(3, 2);
    select_bits_slice = std_slice(3, 1);
    select_bits = std_rsh(3);
  }
  wires {
    port2_select_eq.left = select_bits_slice.out;
    port2_select_eq.right = 1'd1;
    port2_select = port2_select_eq.out;
    port1_select_eq.left = select_bits_slice.out;
    port1_select_eq.right = 1'd0;
    port1_select = port1_select_eq.out;
    shared_addr_slice.in = merged_addr0;
    shared_addr0 = shared_addr_slice.out;
    select_bits_slice.in = select_bits.out;
    select_bits.right = 3'd2;
    select_bits.left = merged_addr0;
  }
}
component mem_0_comp<"toplevel"=1>(@clk clk: 1, port0_addr0: 3, port0_write_data: 32, @go @static(1) port0_write_en: 1, @reset reset: 1) -> (@done port0_write_done: 1) {
  cells {
    merge_hw_0_port_2_write_en_gated = std_and(1);
    merge_hw_0_port_1_write_en_gated = std_and(1);
    merge_hw_0_write_dones_or_d1_i0 = std_or(1);
    merge_hw_0_write_done_port1 = std_wire(1);
    merge_hw_0_write_done_port2 = std_wire(1);
    merge_hw_inst0 = merge_hw_0();
    mem_0_comp_bank_0 = ram_1w(32, 4, 2);
    mem_0_comp_bank_1 = ram_1w(32, 4, 2);
  }
  wires {
    mem_0_comp_bank_0.clk = clk;
    mem_0_comp_bank_1.clk = clk;
    merge_hw_0_port_2_write_en_gated.left = merge_hw_inst0.port2_select;
    merge_hw_0_port_2_write_en_gated.right = port0_write_en;
    merge_hw_0_port_1_write_en_gated.left = merge_hw_inst0.port1_select;
    merge_hw_0_port_1_write_en_gated.right = port0_write_en;
    port0_write_done = merge_hw_0_write_dones_or_d1_i0.out;
    merge_hw_0_write_dones_or_d1_i0.left = merge_hw_0_write_done_port1.out;
    merge_hw_0_write_dones_or_d1_i0.right = merge_hw_0_write_done_port2.out;
    merge_hw_inst0.merged_addr0 = port0_addr0;
    mem_0_comp_bank_0.write_data = port0_write_data;
    mem_0_comp_bank_0.write_en = merge_hw_0_port_2_write_en_gated.out;
    merge_hw_0_write_done_port2.in = mem_0_comp_bank_0.write_done;
    mem_0_comp_bank_0.addr0 = merge_hw_inst0.shared_addr0;
    mem_0_comp_bank_1.write_data = port0_write_data;
    mem_0_comp_bank_1.write_en = merge_hw_0_port_1_write_en_gated.out;
    merge_hw_0_write_done_port1.in = mem_0_comp_bank_1.write_done;
    mem_0_comp_bank_1.addr0 = merge_hw_inst0.shared_addr0;
  }
  control {
  }
}
```

Basically, there is a combinational path from the go/done pairs from the primitive to the top level component. Hence, it should have the same latency. However, the go/done pairs are coming from 2 different instances and then muxed. So this is a weird use case. Maybe instead of erroring out it should just return some conservative solution?

rachitnigam, 2023-05-02:
Okay, I've reduced it to this case:
```
import "primitives/core.futil";

component mem_0_comp<"toplevel"=1>(@go @static(1) go: 1) -> (@done done: 1) {
  cells {
    b0 = std_mem_d1(32, 4, 2);
    b1 = std_mem_d1(32, 4, 2);
  }
  wires {
    b1.write_en = 1'd1;
    b0.write_en = 1'd1;
    done = b1.done | b0.done ? 1'd1;
  }
  control {}
}
```

Definitely looks like a bug!

rachitnigam, 2023-05-02:
Okay, it's even more stupid:
```
component mem_0_comp<"toplevel"=1>(@go @static(1) go: 1) -> (@done done: 1) {
  cells {
  }
  wires {
  }
  control {
  }
}
```

closed 2023-05-02 by rachitnigam

sampsyo, 2023-05-02:
Huge test-case reduction victory!

commit a0281edc (2023-05-02): Fixes #1435 (#1436)