calyxir/calyx #1438: Correctness of Compiling Static While Loops

calebmkim, 2023-05-03:
(This is from a synchronous discussion w/ @paili0628, so @paili0628 feel free to add any clarifications). 

### Counter Example (Sort of)

The following file, [test_static_while.futil.txt](https://github.com/cucapra/calyx/files/11391068/test_static_while.futil.txt), produces different answers for when we compile "normally"/TDCC (it gives us 21 in the memory) vs. when we compile using the technique in #1431 (it gives us 15 in the memory). **However**, this program relies on the fact that undefined ports (e.g., when we read from an undriven combinational component) default to 0 in order to get the result we want. (so this discussion sort of relates to #922). I _think_ that, if there are no undriven combinational cells that are ever being read from, it's impossible to make a counterexample and our compilation in #1431 is fine. 

The basic explanation is this: the while loop is in the form `while l.out { A; B: }` where `l` is a `std_lt` cell. `l` uses the following continuous assignments: `l.right = 5'd6; l.left = w.out`, where `w` is a wire. 

**The key in the program is how `w` is driven**: `w` is driven by `incr.out` when `A` is executing, whereas it is driven by `p.out` anytime `A` is not executing. How do we do this? We add the assignment `w.in = incr.out` to group `A` and then add the continuous assignment `w.in = !A_exec_wire.out ? p.out`. `A_exec_wire` is driven by `1'd1` in `A`, but is _not driven_ anywhere else, meaning that it defaults to 0 everywhere else. Therefore, `w.in` is driven by is driven by `incr.out` while `A` is executing, whereas it is driven by `p.out` anytime `A` is not executing (since `A_exec_wire` will default to 0 anytime `A` doesn't execute), as desired. This means that our trick from #1431 of "checking on the first cycle" doesn't actually work, since it will read from `incr.out` to check the condition, whereas we really should be reading from `p.out` to check the condition. 

### General Takeaway 
In general, it seems there is a difficulty with while loop behavior in Calyx. For example, consider the following while loop: `while port { body}`. It seems, just from looking at this, that `port` should be checked in the time _in between_ the end of the body's execution and the start of the body's next execution. However, if we want to avoid "wasting" a cycle in between the execution of bodies of the while loop, then this "in between" time doesn't actually exist: the last cycle of the body is immediately followed by the first cycle of the body's next execution.

rachitnigam, 2023-05-04:
I haven't looked into the example deeply yet but worth reading our documentation on [undefinedness in presence of undriven reads](https://docs.calyxir.org/lang/undefined.html#undriven-ports)

sampsyo, 2023-05-08:
Wow wow wow; really nice work tracking this down to a concrete counter-example, @paili0628 and @calebmkim! Thank you for the complete code listing and the clear explanation. It's really helpful to keep the issue grounded.

I think your summary at the end is a perfect distillation of the *semantic* question here. "When" should we consider the condition to be evaluated? Or, to put it in terms of the programmer's obligation, when should the program be sure to make the condition defined and relevant?

Let's enumerate the possible answers to this question:

1. The condition is evaluated before the body executes, during a cycle in which the body must not be executing.
2. The condition is evaluated during the first cycle of the body's execution.
3. The condition *may* be evaluated at *any time* during or before the body's execution, so the body has an obligation to keep the condition value stable throughout its entire duration.
4. The condition is evaluated at some point before the first cycle of the body's execution finishes. But the body is not guaranteed to have started execution when the condition is evaluated. So the body has an obligation to avoid changing the condition's value *during its first cycle*. (Updating a register in the first cycle, and hence affecting the register's value in the *second* cycle, is fine.)

(I hope I am missing other possibilities, because none of these are 100% satisfying.)

Options 1 and 2 are meant to be direct reflections of our implementation strategies, for TDCC and this new compiler pass respectively. I don't like those options very much because they are really specific about exactly when the condition must be evaluated, which constrains how the compiler must work. What we want, I argue, is a semantic definition that allows *either* implementation to be correct—one that comes with a well-formedness "law" that disqualifies programs that can tell the difference between the two strategies.

Option 3 is tempting, but I don't think it works. Say your condition is driven by a register (the common case); you want to be able to update that register at any point in the loop's body. This policy would require loops to leave the register alone *until the very end of the body*, when they're finally allowed to make that exit decision.

Option 4 is a straightforward attempt to unify options 1 & 2. Basically, it says "the compiler's policy may be option 1 and it may be option 2; the program must be written carefully to avoid telling the difference." This is OK, but I don't love the fact that it still "thinks" in terms of clock cycles. The rest of Calyx's semantics (static stuff aside) don't say anything about individual clock cycles, just what happens before, after, or during other stuff.

Another category of approach could be to define the semantics differently for `while` loops with dynamic bodies vs. `while` loops with static bodies. That is a little bit unsatisfying too because we like the property where "static erasure" preserves program semantics. But perhaps we can still preserve that property, in the sense that the well-formedness requirements on `while`-static loops are engineered to be stricter than those on `while`-dynamic loops.

rachitnigam, 2023-05-09:
One possible answer to the question: "can we have undriven signals in this program" comes from this issue: https://github.com/cucapra/calyx/issues/1169

One of the conditions for something to be a `@data` component is that it doesn't affect a cell that shows up as a port for `if` or `while`. I think another way to parse this constraint is that ports in those locations are probably never allowed to take undefined values since that would mean that the guard for everything under the `if` might be triggered? 

One way to think about undriven ports is that they are allowed to take any value they want so perhaps that's the reason this program should be ruled out

rachitnigam, 2023-05-21:
@calebmkim implemented the requirement that conditional ports are `@stable` (https://github.com/cucapra/calyx/pull/1467) and @paili0628 implemented the proposed compilation scheme for unbounded loops (#1431)

closed 2023-05-21 by rachitnigam