calyxir/calyx #1448: MrXL: less boilerplate in .data files

anshumanmohan, 2023-05-08:
MrXL data files all have the same "format" field. It would be nice to have a `fud` pass that just tacks the format on, allowing the user to write cleaner code.

For example, see [dot.mrxl.data](https://github.com/cucapra/calyx/blob/master/frontends/mrxl/test/dot.mrxl.data):
```
{
    "avec_b0": {
        "data": [1, 2, 3, 4],
        "format": {
            "numeric_type": "bitnum",
            "is_signed": false,
            "width": 32
        }
    },
    "bvec_b0": {
        "data": [1, 2, 3, 4],
        "format": {
            "numeric_type": "bitnum",
            "is_signed": false,
            "width": 32
        }
    },
    "dot_b0": {
        "data": [
            0, 0, 0, 0
        ],
        "format": {
            "numeric_type": "bitnum",
            "is_signed": false,
            "width": 32
        }
    }
}
```

I'd love to be able to write
```
{
    "avec_b0": [1, 2, 3, 4],
    "bvec_b0": [1, 2, 3, 4],
    "dot_b0" : [0, 0, 0, 0]
}
```
And then have `fud` convert it in the obvious way into the above.

anshumanmohan, 2023-05-10:
@susan-garry I hope you don't mind that I've assigned this to you! I figure you'll become a fud-ninja shortly and this will be quick work for you :)

sampsyo, 2023-05-11:
Ah, I just added a comment in https://github.com/cucapra/calyx/issues/1450#issuecomment-1544700210 that came to more or less the same conclusion. In fact, I'm not entirely clear on the difference between this issue and that one—maybe they are the same?

anshumanmohan, 2023-05-11:
Hehe I guess I had split a hair by having them be separate issues. I'll close this one, since I agree that your comment in #1450 summarizes this stuff nicely.

closed 2023-05-11 by anshumanmohan