calyxir/calyx #1450: MrXL: autobank

anshumanmohan, 2023-05-08:
Write a `fud` pass that automates banking.

When writing a program of the form
```
input foo: int[4]
...
```
MrXL requires a data file of the form
```
{
  "foo_b0": {
    "data": [
      1,
      2
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  },
  "foo_b1": {
    "data": [
      3,
      4
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
  }
  ...
}
```

That is, it requires manual banking of the logical `foo` into two chunks. It would be great if I could pass a data file of the form 

```
{
  "foo": {
    "data": [
      1,
      2,
      3,
      4
    ],
    "format": {
      "numeric_type": "bitnum",
      "is_signed": false,
      "width": 32
    }
    "banking": 2    
  }
  ...
}
```
and have a `fud` pass that converts the latter into the former.

sampsyo, 2023-05-11:
If I can refine this proposal one step farther: the goal here is to make JSON data files that MrXL wants to take as input "natively," which are distinct from the Calyx-formatted JSON files. The new MrXL-native data files should:

* Not be banked. That is, if the MrXL source code declares an array called `a`, then there is just one thing called `a` in the MrXL data file.
* Use MrXL types (not Calyx types), if at all. Like, the declaration in MrXL looks like `input a: int[4]`; if we need to write the type down in the input file, let's just use `int` for that instead of the Calyx-style `"format": { "numeric_type": "bitnum", "is_signed": false, "width": 32 }`.

Then, the MrXL tooling (whether it's a fud stage or not!) can be responsible for translating this MrXL format into the Calyx-native format.

There is one choice to make:

1. The banking and type information could be contained within the MrXL JSON data file, as @anshumanmohan proposes above.
2. The banking and type information could come from the MrXL source code. Like, the data file could just look like `{"a": [1,2,34], "b": [...]}` with no banking or type info at all. When we do the JSON-to-JSON translation, then, we also have to look at the MrXL source code to know how many banks to produce and which Calyx types to generate.

I would suggest that we want option 2 if it doesn't seem like too much effort.

anshumanmohan, 2023-05-11:
#1448 is now absorbed into this. And your option 2 sounds like the ticket indeed; thanks for that!

anshumanmohan, 2023-05-13:
I'll add another nice-to-have to this issue: would be great if MrXL-native JSONs could just punt on the `output` registers/arrays. 

For the MrXL program
```
input avec: int[4]
output sos: int
squares := map 2 (a <- avec) { a * a }
sos := reduce 1 (acc, i <- squares) 0 { acc + i }
```
a reasonable MrXL-native .data file would be
```
{ "avec": [1,2,3,4] }
```
The MrXL source code has enough info to infer that the "real" (i.e. Calyx-native) .data file needs 
1. banking
2. the addition of `format` blocks
3. the addition of a register called "sos".

anshumanmohan, 2023-05-15:
Summarizing after the meeting we just had:
- [ ] Reduce `format`-based boilerplate.
- [ ] Autobank, reading the necessary info from the MrXL source program.
- [ ] Generate the memories required for `output`, again by reading the MrXL source program.