calyxir/calyx #1469: Automate process of releasing new crates and docker image

rachitnigam, 2023-05-12:
We should automate the process of releasing new cargo crates and docker image for Calyx using Github's "release" feature. When a new release is created, a github action can run and release new version of packages and the docker container based on the version of the release.

One challenge is separating out the release of crates from the docker image since the image might change more often than the crates.

rachitnigam, 2023-05-24:
Documenting a failed attempt to do this in #1502. We tried to automate the release of containers on two platforms: `x86` and `arm` but the arm runner times out on github actions (6 hours) since it is being emulated by QEMU. Building the whole Calyx container seems hopeless: it couldn't even fully build Verilator. It should be possible to setup cross-compilation scripts and run most of the build process directly on x86 but it is more work than I signed up for. The following job, that attempted to release a `v0.3.0` container documents this timeout: https://github.com/cucapra/calyx/actions/runs/5058160411

sampsyo, 2023-05-24:
Awesome. It was worth a shot—too bad that the emulated ARM build was so disastrously slow.

On a lark, I'm going to email [Depot, a startup that does Docker builds as a service](https://depot.dev/docs/guides/arm-containers) to ask about academic discounts.

rachitnigam, 2023-06-27:
@sampsyo should we close this issue since you implemented the depot flow? I guess the only other thing to do is make sure that depot releases tagged versions correctly.

sampsyo, 2023-06-27:
Yes, I think we are all set! It _should_ suffice to `git tag vX.Y.Z ; git push --tags`, but I guess we'll see the next time we do it.

closed 2023-06-27 by sampsyo