calyxir/calyx #1472: MrXL: allow different commands to have different banking factors

anshumanmohan, 2023-05-13:
This would have been needed on the way to #1414 anyway, but I'd like it sooner for tutorial purposes (#1457).  

I currently cannot compile the following sum of squares program:
```
input avec: int[4]
output sos: int[4]
squares := map 2 (a <- avec) { a * a }
sos := reduce 1 (acc, i <- squares) 0 { acc + i }
```
It interprets fine, but `mrxl test/sos.mrxl` gives 
```
...
Exception: Previous use of `squares` had banking factor 2 but current use has banking factor 1
```
Increasing the parallelism factor of `reduce` is not an option, since that is not implemented. This means I'm left with 
```
...
squares := map 1 (a <- avec) { a * a }
sos := reduce 1 (acc, i <- squares) 0 { acc + i }
```
which is a much weaker program to work with pedagocially.

anshumanmohan, 2023-05-15:
I gave it some more thought. I think an array needs to be banked per the _[LCM](https://en.wikipedia.org/wiki/Least_common_multiple) of the various banking factors that ever bank the array_. For example, 

```
input avec: int[24]
squares := map 4 (a <- avec) { a * a }
add5 := map 3 (a <- avec) { a + 5 }
...
```

- In the data file we need to convert `avec` into 12 banks, `avec_b0` through `avec_b11`. 
- When `squares` is being computed, we sorta re-group those 12 banks into 4, not via any actual re-banking but just be considering `avec_b0`, `avec_b1`, and `avec_b2`, to be contiguous, `avec_b3`, `avec_b4`, and `avec_b5`, to be contiguous, and so on.
- Similarly, when `add5` is being computed, we re-group into 3 banks.

I've laid out the general case, but in practice array-lengths and banking factors will probably all be powers of 2, so this will be easier: just banking per the _finest_ banking factor, and then re-group as necessary.

anshumanmohan, 2023-05-15:
Done with 1 and 2 above. The PR is once again ready for review.

anshumanmohan, 2023-05-18:
I'm taking this off the tutorial milestone. The new milestone is #1474. I will incorporate some of the discussion above into my educational slide/handout. 

Basically we want them to try to implement the feature that this issue talks about, and it's not critical that we have the whole answer waiting on the side.

anshumanmohan, 2023-08-15:
@rachitnigam should we just close this as a wontfix? Or maybe convert it into a discussion or somesuch?

rachitnigam, 2023-08-16:
Maybe the "Low Priority" tag is in order? The task does seem pretty actionable but probably not critical for anyone's use. However, I think in the past @sampsyo has voted in favor keeping issues like this open as a way to onboard new people.

sampsyo, 2023-08-16:
A "low priority" label sounds about right to me!

anshumanmohan, 2023-08-16:
Done deal!