calyxir/calyx #1479: Tutorial: assets needed

anshumanmohan, 2023-05-16:
This is a tracking issue for all the stuff we need at hand for the tutorial. I've added some provisional assignments, but let's discuss!

- [ ] Intro, setup, first Calyx program @rachitnigam
- [ ] Pollen @susan-garry
- [ ] Cider @EclecticGriffin 
- [ ] `fud` (GB/PS/RN?)
- [ ] Educational materials for the competition @anshumanmohan 
- Memory banking
    - Slides explaining what MrXL is already doing
- Reduction trees
    - Toy solution available on the side
    - Slides explaining what’s going on, pipelining as a possibility, etc
- Toy arbitration logic
    - Toy solution available on the side
    - Plus slides explaining what’s going on, and how this applies letting the same memory be banked with different parallelism factors

anshumanmohan, 2023-05-20:
[Here](https://www.icloud.com/keynote/074tC5fuawyy4GQLUdCOV6IdA#red%5Ftrees) are some slides as a primer for reduction trees. Seeking feedback at any level! One flaw is that folks _need_ the animation right now; the slides viewed statically all look the same. I'll add "Day {i}" annotations to the matches. I just wanted your thoughts before getting too deep into it.

anshumanmohan, 2023-05-21:
And [here](https://www.icloud.com/keynote/093n5xPkBNJoS26zentQwE-_w#banking%5Fand%5Farbitrage) are further slides, this time explaining memory banking and arbitrage. Same deal: I have ideas about how to make these less reliant on animation, but I'm looking for general commentary before going too deep.

rachitnigam, 2023-05-21:
Thanks for getting us started @anshumanmohan! Just as a suggestion for the rest of us, maybe we should just have one slide deck and create "sections" in it for our various talks? It would centralize our efforts and simplify things!

sampsyo, 2023-05-22:
It seems 100% fine to rely on animation for the slides, FWIW! If people want to view them "statically" for reference, you can export to PDF from Keynote with the "print every stage of builds" option enabled.

anshumanmohan, 2023-05-23:
Update, I've mashed my slides into one .key. I've changed the link and the text above

anshumanmohan, 2023-05-23:
@rachitnigam, reminder to change the [schedule](https://calyxir.org/tutorial/) to give Cider 5 mins

anshumanmohan, 2023-05-23:
In view of the changes I've made to the frontend tutorial, I propose a change to the schedule.

Right now we have
10am-10.05am | MrXL: A map-reduce frontend
10.05am-11am | Compiling MrXL to Calyx

And I think we could instead have
10am-10.15am | MrXL: A map-reduce frontend
10.15am-11am | Add your own extensions to MrXL

That is, in the first 15 mins, we'll walk them through the tutorial as written. We will gloss over the parts where the tutorial shows how to compile MrXL using the eDSL, since that's really best for working through yourself. Someone lecturing that same stuff at you won't help.

Then we'll leave 45 mins for them to add one of the "warmup" extensions to MrXL. This will put them in a good place to add a real extension later on.

anshumanmohan, 2023-05-23:
Speaking of later on, I propose that we change

> 11.25am-12.25pm | Hardware Performance 101 / MrXL competition

into three tracks.

1. Reduction trees, for those interested in speedup. 
2. Arbitration and multi-banking, for those interested in that.
3. Office hours, for those just wanting to backfill about what the heck is going on.

anshumanmohan, 2023-05-23:
I dry-ran the tutorial exactly as written, with the exception that I skipped to the not-yet-merged branch [mrxl-input-dynamic](https://github.com/cucapra/calyx/tree/mrxl-input-dynamic) when it came time to do MrXL stuff.

I have flagged my issues here: https://github.com/cucapra/calyx/issues/1508. After those issues are fixed, someone else should do a dry run.

priyasrikumar, 2023-05-23:
I can own fixing those issues! Most of them seem doable but will ping others if there are any issues with resolving those. I will also do the tutorial dry run after that and see if there are any other hiccups!!

rachitnigam, 2023-06-13:
Should we consider this completed?

anshumanmohan, 2023-06-13:
Yup!

closed 2023-06-13 by anshumanmohan