calyxir/calyx #1481: Tutorial: create opportunity for a bug that Cider then fixes

anshumanmohan, 2023-05-17:
We've decided that we'll have a 5-min demo of Cider. It would be great if we could somehow tee this up for Griffin.

If we cannot, not the end of the world: Griffin clearly has an example at hand; they showed it to Owolabi during their A exam today!

anshumanmohan, 2023-05-18:
Let's talk about this on Monday, @rachitnigam  and @susan-garry. I'm not totally sure how we we'll get `mrxl/gen_calyx` to introduce this bug as a one-off without being hella buggy all the time. 

Maybe we could generate `sos.calyx`:
```
mrxl test/sos.mrxl > test/sos.calyx
```

and then doctor `sos.calyx` by hand to have the bug that Griffin presented in their A-exam:
```
while (idx < 4) {
   ...
}
```
but idx has been declared as width-2 and so it overflows after 3, never equaling 4.

rachitnigam, 2023-05-19:
Yeah, that’s what I meant when I proposed having a bug to show off

anshumanmohan, 2023-05-24:
Update after today's meeting: Griffin will do something up, not linked to MrXL!

anshumanmohan, 2023-06-09:
This is no longer an issue that needs to be closed via an actual code push, so I'm closing it. The need for the example is captured by the need for a Cider talk :)

closed 2023-06-09 by anshumanmohan