calyxir/calyx #1496: Producer consumer example using `ref` cells

rachitnigam, 2023-05-21:
@anshumanmohan recent work on the arbitration component in Calyx (#1474) shows off the expressive power of `ref` cells. I think generally, it's a feature whose consequences are pretty unexplored and, while technically simple to implement, I think it really transforms the language in the significant manner.

One cool showcase of this can be an implementation of a producer-consumer example that pass a common `fifo` using `ref` cells.

Here is a fake implementation
```
primitive std_fifo[WIDTH, SIZE](
    @go(1) pop_en: 1,
    @go(2) push_en: 1,
    in: WIDTH,
) -> (
    @done(1) pop_done: 1
    @done(2) push_done: 1,
) { 
    /* Implementation */
}

component Producer() -> () {
  cells {
    ref fifo = std_fifo(32, 10);
    r = std_reg(32); // Contains value from complex computation
  }
  wires {
    group write_fifo {
        fifo.push_en = 1'd1;
        fifo.in = r.out;
        write_fifo[done] = fifo.push_done;
    }
  }
  control {
    // <perform some computation>
    // Push computation to FIFO
    write_fifo;
  }
}

component Consumer() -> () {
    cells {
        ref fifo = std_fifo(32, 10);
        r = std_reg(32); // Read the value from FIFO
    }
    wires {
        group read_fifo {
            fifo.pop_en = 1'd1;
            r.in = fifo.out;
            r.write_en = fifo.pop_done;
            read_fifo[done] = r.done;
        }
    }
    control {
        read_fifo;
        // <Use the value from the fifo>
    }
}

component main() -> () {
    cells {
        p = Producer();
        c = Consumer();
        fifo = std_fifo(32, 10);
        cond = std_reg(1); // Done condition of the program
    }
    wires {}
    control {
        while cond.out {
            // Execute producer and consumer in parallel
            par {
                invoke p[fifo=fifo]()();
                invoke c[fifo=fifo]()();
            }
        }
    }
}
```

We can probably implement this pretty easily. @paili0628 you might be interested in thinking about this since dynamic FIFOs are truly a case that can only be supported using dynamic scheduling while the code in the producer and consumer can be static.

anshumanmohan, 2023-05-21:
Love it! And, in a few months... [PIFO trees](https://arxiv.org/pdf/2211.11659.pdf)! 😈

rachitnigam, 2023-05-22:
@anshumanmohan cool idea! I think it actually wouldn't be too hard, especially if we build the interface for `std_fifo` correctly. That is as long as PIFOs expose `push` and `pop` "methods", they can be used in the same places as FIFOs!

sampsyo, 2023-05-22:
Super cool idea. The `push_en`/`pop_en` interface does make me wish once again for multiple go/done "methods" so that the `std_fifo` component could be `invoke`able… but that is totally tangential to the matter at hand.

rachitnigam, 2023-05-22:
Yup, definitely a good use case for #986! We just need a good syntax before implementing it :-) (It's not technically challenging at all)

anshumanmohan, 2023-06-14:
Heyo @paili0628, I was just wondering if you'd mind if I took a crack at this and ran what I have past you. I have a long flight coming up :)

rachitnigam, 2023-06-14:
The [`std_sync_reg`](https://github.com/cucapra/calyx/blob/master/primitives/sync.sv#L13) implementation might be of use when implementing a FIFO in Verilog. `std_sync_reg` is a one cell, multi-reader, multi-writer FIFO. FIFOs in general can get tricky because of the concurrent reads and writes so I recommend choosing a simple design/using something from the internet.

rachitnigam, 2023-06-14:
(Also not sure if @paili0628 is explicitly assigned to this right now since she's working on a bunch of other things right now)

anshumanmohan, 2023-06-22:
I'm going to take this on, with help from all of you and especially @paili0628! 

The plan is:
1. FIFOs.
2. PIFOs built using FIFOs. A PIFO is just a priority queue with the additional property that rank-ties are broken using "first come, first serve" order.
3. PIFO trees built using PIFOs, specifically the motivating example shown in §2 of [this paper](https://arxiv.org/pdf/2211.11659.pdf).

I'll go over this plan backwards, since that is roughly increasing order of complexity.

## PIFO trees built using PIFOs
I think this will be fine, given the right kind of PIFO. The main thing is that our motivating example guides us towards a critical requirement from our PIFOs: I think they need to be able to assign _arbitrary_ ranks, not just "hi"/"lo" ranks. I tried it out using just hi/lo, but it turns out that if you simplify the "mission" in the motivating example from the paper into something like:
> Strictly prefer $B$ packets to $R$ packets. $R$ packets are of two flavors, $T$ and $P$, and you must strictly prefer $T$ packets to $P$ packets.

we would fall into a degenerate case and no longer need PIFO trees. I can sketch it out fully if folks are interested. 

## PIFOs built using FIFOs

### A PIFO with exactly $n$ ranks, for some known $n$.
_This won't work for our purposes, but I'll sketch it out._

Say we want a PIFO with maximum capacity 10. Say we know beforehand that the PIFO will only ever assign $n$ ranks. For this example let's say $n=2$ and let's call the ranks "hi" and "lo", but this generalizes to any $n$ straightforwardly.

Just create two FIFOs, each of capacity 10. The first, called $FIFO_1$, is for "hi" values and the second, $FIFO_2$, is for "lo" values.
- $len(PIFO) = len(FIFO_1) + len(FIFO_2)$
- $push(v, r, PIFO)$: if $len(PIFO) = 10$, raise an "overflow" flag and exit. Otherwise, the charge is to enqueue value $v$ with rank $r$, and $r$ better be either $1$ or $2$. Enqueue $v$ into $FIFO_r$. Note that the FIFO's enqueue method is itself partial: it may raise "overflow", in which case we propagate the overflow flag.
- $pop(PIFO)$: if $len(PIFO)$ = 0, raise an "underflow" flag and exit. Try $pop(FIFO_1)$. If it succeeds it will return a value $v$; just return $v$. If it fails because of underflow, return $pop(FIFO_2)$.

### A PIFO that achieves a 50/50 split between two flows.
_This is what I actually need. It's not hard to see why it doesn't work using the approach above. I'm sketching a possible solution below, but I'm much less confident in this._

Up to the availability of values, this PIFO seeks to alternate between its two flows. Its maximum capacity is 10. Create $n$ FIFOs, as many FIFOs as there are flows, each of capacity 10. Let's say there are two flows, $R$ and $B$, and our FIFOs are called $FIFO_R$ and $FIFO_B$. Maintain additionally some kind of register in your state that points to which of these FIFOs is "hot".

- $len(PIFO) = len(FIFO_R) + len(FIFO_B)$
- $push(v, f, PIFO)$: if $len(PIFO) = 10$, raise an "overflow" flag and exit. Otherwise, the charge is to enqueue value $v$, that is known to be from flow $f$, and $f$ better be either $R$ or $B$. Enqueue $v$ into $FIFO_f$. Note that the FIFO's enqueue method is itself partial: it may raise "overflow", in which case we propagate the overflow flag.
- $pop(PIFO)$: if $len(PIFO)$ = 0, raise an "underflow" flag and exit. Try $pop(FIFO_{hot})$. If it succeeds it will return a value $v$; just return $v$. If it fails because of underflow, return $pop(FIFO_{nothot})$, where $nothot$ is just the FIFO not pointed to by $hot$; it is easily inferred because we only have two FIFOs. In either case, flip $hot$ to point to the other FIFO.

## FIFOs

The exposition above has shown us a little more about what we need from our real workhorses, the FIFOs. 

In addition to Rachit's very helpful skeleton [above](https://github.com/cucapra/calyx/issues/1496#issue-1718563222), I think we need overflow and underflow flags that say, basically, "I'm done, but something went wrong and you should not read my out value."

In terms of the actual implementation, I was thinking of a memory of length 10 (for a FIFO with capacity 10), along with pointers to tell us where to push to and where to pop from. These pointers will be clever enough to simulate a circular queue. For instance, after ten pushes and one pop, the queue may look like 

|   | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 
|---|---|---|---|---|---|---|---|---|---| 

but it is not full; it has room for an incoming $11$ as so:
| 11 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 
|---|---|---|---|---|---|---|---|---|---|

anshumanmohan, 2023-06-27:
Update after meeting: I will try this as proposed above, i.e:
- Single-producer single-consumer FIFOs built using circular queues, as above. No `sync` needed.
- PIFOs set up to do round-robin, as above.
- PIFO trees set up to arrive at the example in the paper.
 
This will be the first step, and will feature no static timing or pipelining. The next step will be static timing, and the step after that will be pipelining.

anshumanmohan, 2023-07-02:
I'm realizing I have a bug. Above, I say
> Up to the availability of values, this PIFO seeks to alternate between its two flows. I say "up to availability" because, if one flow is silent and the other is active, the active ones gets to emit consecutive values (in temporary violation of the 50/50 rule) until the silent flow starts transmitting up again. At that point we go back to 50/50, **_not_ giving the hitherto-silent flow any "credit" for the time that it was silent.**

The very last bit of the spec is incorrect: the hitherto-silent queue _does_ get credit for the time it was silent. In particular, it gets to cut to the front of the unbalanced part of the queue and gets matched with the first unmatched candidate from the other queue. This is what the three possible queues on page 3 of [the paper](https://arxiv.org/pdf/2211.11659.pdf) are saying.

I'll keep thinking, but I just wanted to flag it. We'll need to something slightly fancier with FIFOs to get this new PIFO spec. Whatever it is, it's a little more subtle than the "hot/not" thing I have sketched above.

closed 2023-08-15 by anshumanmohan