calyxir/calyx #1509: Minor expansion to CI testing

anshumanmohan, 2023-05-23:
@rachitnigam wanted the new arbitration logic example to go into top-level CI testing. However, it looks to me like none of the other `calyx-py` code is tested at every push. `.expect` files to exist, but they have Calyx code and not Verilog outputs of running that Calyx code. This makes sense; they are testing the builder.
 
For now I have done in Rome as the Romans do: `reduction_tree.py` and `arbiter_6.py` live under `calyx-py/` and are accompanied by .expect files that contain Calyx code. I have checked that I can run this Calyx code correctly with Verilog.

Let's talk about what you'd like, and where you'd like it to live!

rachitnigam, 2023-05-24:
Huh, weird! We should enable them again. The woes of having multiple `runt.toml` files :-(

anshumanmohan, 2023-05-24:
Well, lemme know if I've misdiagnosed the situation! The top-level runt.yml mentions calyx-py only twice, and it seems to me that those are both about installing it. https://github.com/cucapra/calyx/blob/04fd154c7836ad8354fa70cd9f34194b3c801794/.github/workflows/rust.yml#L92-L95

rachitnigam, 2023-05-24:
Ah, the `runt.toml` configuration does mention it but only insofar as *generating* calyx from the python but does not run them.

PR #1514 (2023-05-24): `calyx-py` testing
Fixes #1509.
closed 2023-05-24 by anshumanmohan