calyxir/calyx #1513: Tutorial Checklist

rachitnigam, 2023-05-24:
Tracking the low-level tasks that need to be done before the tutorial:
- [ ] Dry-run
  - [ ] `x86`
  - [ ] M1 mac
- [ ] Docker Issues
  - [ ] Make sure people can `docker pull` the GHCR image
  - [ ] Ensure that `fud check` reports that things are correctly installed
  - [ ] Make sure that `fud` commands within the docker container work outside the `calyx` repo (no relative paths)

## Final Checklist
- [ ] Release latest version of docker containers for `x86` and `arm`
- [ ] Check that we have audio setup for virtual speakers
- [ ] Proof-read the documentation and fix any broken links

anshumanmohan, 2023-05-24:
Should the Docker have `vim` or some other text editor installed?

anshumanmohan, 2023-06-08:
I think I was the only person who could not pull Docker, and now I can too. Weird. Sorry about the stress. Checking that box.

rachitnigam, 2023-06-14:
Argh, sorry we missed this. How about we put up a note on the slide saying "please install and configure your favorite editor" once people install the docker image?

rachitnigam, 2023-06-14:
The website now has a link the PDF (right before the schedule). Once someone verifies the link, we can close this issue

anshumanmohan, 2023-06-14:
Can confirm that the links works

anshumanmohan, 2023-06-14:
Re the text editor:

A little odd to be sure, but yeah this may be the band-aid solution. To make it a slightly nicer band-aid, you could give them the necessary `apt` incantations on the slides. May want to check that they work (e.g. that `sudo` is not required...)

Just to be clear, since we're closing in on this: cool if this gets added to your list as you make the initial slides?

closed 2023-06-27 by rachitnigam