calyxir/calyx #1517: Builder: `par` generation is non-deterministic

rachitnigam, 2023-05-24:
@EclecticGriffin we talked about this but not sure if this was already addressed since #1503 seems to run into this.

EclecticGriffin, 2023-05-24:
This is fixed if you use the `par(...)` constructor over the set literal. We may want to deprecate/remove the set literal construction

rachitnigam, 2023-05-24:
Let's remove it. The current docs use it and it's not correct: https://docs.calyxir.org/calyx-py.html

Also, CC @priyasrikumar to change this

EclecticGriffin, 2023-05-24:
see: https://github.com/cucapra/calyx/pull/1503/files#r1204378462

sampsyo, 2023-05-24:
It's probably a good idea to remove the set trick! It was too cute to live. Especially since @EclecticGriffin wisely added an explicit `par`, as referenced above.

PR #1522 (2023-05-24): [calyx-py] Minor bulider tweaks
- per @sampsyo's comment, changed `sub_component` to `comp_instance`
- removed the set literal construction and replaced it with a `TypeError`
- additional docstrings and value checks

Closes #1517
closed 2023-05-24 by sampsyo