calyxir/calyx #1532: Dry run thoughts!

priyasrikumar, 2023-05-29:
I just did a dry run of the tutorial, and wanted to share some notes I had here!

- The `ROOT_INSTALL` issue is still present (see [#1508](https://github.com/cucapra/calyx/issues/1508#issuecomment-1560069195)).
- There were a few small documentation fixes, addressed in #1531 

Some minor nits with the frontend tutorial documentation: 
- [ ] We [direct](https://github.com/cucapra/calyx/blob/master/docs/tutorial/frontend-tut.md?plain=1#L12) readers to go to `frontends/mrxl`, but in order to run the example, they need to run a command from the root of the Calyx repository, and they go to `frontends/mrxl` in the next line. Maybe we remove it?
- [ ] [This command](https://github.com/cucapra/calyx/blob/master/docs/tutorial/frontend-tut.md?plain=1#L87-L89) has comments next to it which made it finicky to copy, should we maybe make the explanation follow or precede the command?
- [ ] [This command](https://github.com/cucapra/calyx/blob/master/docs/tutorial/frontend-tut.md?plain=1#L290) and the next two after it might only need to reference the file by `test/squares.mrxl`? Since folks will be in the `frontends/mrxl` directory? 

^ Happy to make any of the above changes folks agree on! We can check the ones we want! 

Overall everything went very smoothly and these are just a few minor remarks, so this seems to be in good shape!

priyasrikumar, 2023-05-30:
@anshumanmohan would love to hear your thoughts on the frontend tutorial edits when you have time! truly no rush :)

anshumanmohan, 2023-05-30:
Please push on without me; I don't want to block. From your summary above, this all sounds good to me

rachitnigam, 2023-05-31:
@priyasrikumar The change related to comments sound good. I think there was a rewriting conflict with the latter stuff: @anshumanmohan wrote the docs assuming people where in `frontends/mrxl` and I wrote them people are in the repository root.

All things equal, I prefer to keep instructions tagged to the repository root because it seems like a good "default" place that people can go back to and have things work out. Either way, I don't have a strong opinion on this except that we should be consistent about one choice or the other.

anshumanmohan, 2023-05-31:
Thanks for the catch; let's move to the consistent version for sure!

rachitnigam, 2023-06-05:
@priyasrikumar should we close this?

closed 2023-06-05 by priyasrikumar