calyxir/calyx #1533: the trait `Default` is not implemented for `T`

linuxlonelyeagle, 2023-05-30:
I found that when I run cargo build it gives me an error.I hope someone can help me.
```
error[E0277]: the trait bound `T: Default` is not satisfied
   --> calyx-ir/src/guard.rs:183:29
    |
183 |         let old = mem::take(self);
    |                   --------- ^^^^ the trait `Default` is not implemented for `T`
    |                   |
    |                   required by a bound introduced by this call
    |
note: required because of the requirements on the impl of `Default` for `guard::Guard<T>`
   --> calyx-ir/src/guard.rs:35:24
    |
35  | #[derive(Debug, Clone, Default)]
    |                        ^^^^^^^
note: required by a bound in `std::mem::take`
    = note: this error originates in the derive macro `Default` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider restricting type parameter `T`
    |
145 | impl<T: std::default::Default> Guard<T> {
    |       +++++++++++++++++++++++

For more information about this error, try `rustc --explain E0277`.
error: could not compile `calyx-ir` due to previous error
warning: build failed, waiting for other jobs to finish...
```

rachitnigam, 2023-05-30:
Can you provide steps to reproduce the error? Building from the latest version of the master branch succeeds so not sure how you’re getting this error.

linuxlonelyeagle, 2023-05-30:
I am interested in this [issue](https://github.com/llvm/circt/issues/1679) and even though it has been half a year since it was raised, I still want to complete it. In the past six months, I have learned a lot about MLIR and systolic array, and because I am now interested in Circt, I am planning to re-learn Calyx and also hoping to complete this issue.

rachitnigam, 2023-05-30:
Can you update the version of the Rust compiler you're using (we're using 1.69). The compiler is [continuously tested on every commit](https://github.com/cucapra/calyx/actions/runs/5106195308/jobs/9178249341) so we're certain that it can be built.

If all else fails, you can try using [our docker container](https://github.com/cucapra/calyx/pkgs/container/calyx).

linuxlonelyeagle, 2023-05-30:
> Can you update the version of the Rust compiler you're using (we're using 1.69). The compiler is [continuously tested on every [...]

Thanks!I have resolved the issue by using Rust 1.69, and I will work on the [issues](https://github.com/llvm/circt/issues/1679) listed in the repository. If I have any questions, I will ask for your help.

closed 2023-05-30 by linuxlonelyeagle