calyxir/calyx #1537: Selective Promotion and De-Promotion of Static Groups/Control

calebmkim, 2023-05-31:
Static promotion of groups can present some challenges. 

For example, sometimes, we might promote many, but not all, groups in a `seq`, resulting in something like `seq { static_group; .. // many static groups ... dynamic_group}`. This means we do not promote and therefore keep it as a regular seq. The result is _very_ wasteful hardware, since each `static_group` in the seq will have its own fsm. 

It seems that for these "partially static" control operators, there's two options: 
1) De-Promotion. Do not promote any of the static groups. So we just get something like this: `seq { dynamic_group; .. // many dynamic groups ... dynamic_group}` ( a quick note: this will be difficult/impossible to do if the static-ness is not inferred, i.e., the input calyx is a static group, since we won't really have a done condition to go on for the group). 
2)  Partial control promotion. Instead, we could promote the portion of the seq that we can into a static seq, like so: `seq { static seq { // contains the static groups };  dynamic_group}`

It seems like there will be some sort of tradeoff between the decreased cycle count that we would get from the control promotion and the increased hardware that it generates. Even more broadly, there also seems to be a "cost-heuristic" problem with promoting things like `if` statements as well. We could have something like: `if cond { static<5> seq } else { static<6> seq }` where the branches are slightly different lengths... it might be worth turning it into a static if statement (with the latency being the longer branch) if the difference is small enough).

sampsyo, 2023-05-31:
Awesome; thanks for writing this up. When I was thinking about it a little more, I realized the same thing as you: "demotion" seems impossible in general, so it's probably not a good idea to go that route. (We would need to somehow leave breadcrumbs that says "I automatically promoted this from static to dynamic, but the program will not break if it's demoted again," which seems very tricky semantically.) So it kinda seems like option 2 would be much nicer.

calebmkim, 2023-05-31:
Yeah, I was thinking about this problem some more, and I had an idea (although maybe not a good one): 

We could break up the control promotion into two stages. The first is attaching the `@static` attribute wherever possible. The second is then analyzing where we have placed the `@static` attribute and decide whether we want to "demote" it (i.e., literally just do nothing to it except removing the attribute) or "promote it" to actual `static control` (e.g., `static<10> seq`). 

Of course, this won't help if the Calyx design given to us includes these annoying "partially static" control blocks (unless we tell frontends that we actually prefer them to give us `@static` annotations so that the Calyx compiler can determine whether it's smart to actually use it).

sampsyo, 2023-06-01:
Yeah, that could definitely work! It would have the benefit of decoupling the two steps, but the added complexity of having another attribute whose semantics we would need to define. And we may want to rename the attribute, like, `@maybe_static` or `@static_hint` or something to clarify what it means. That is, it means that the group/statement/whatever is still very much dynamic but some frontend/pass believes that it _might_ be a good idea to convert it to static.

sampsyo, 2023-06-02:
Got to say, it is AWESOME to be able to get immediate feedback like this for a proposed change! SUPER cool to avoid needing to guess.

closed 2023-06-15 by paili0628