calyxir/calyx #1546: Tutorial: changes to the schedule

anshumanmohan, 2023-06-05:
Just gathering up the changes I proposed to the [schedule](https://calyxir.org/tutorial/). The original comments are [here](https://github.com/cucapra/calyx/issues/1479#issuecomment-1559969787) and below. These have been updated in light of the big meeting we had after I posted those comments.

1. Give Cider 5 mins.

2. Right now we have
      > 10am-10.05am | MrXL: A map-reduce frontend
      > 10.05am-11am | Compiling MrXL to Calyx
      
      And I think we could instead have
      > 10am-10.15am | MrXL: A map-reduce frontend
      > 10.15am-11am | Adding `map` to MrXL.
      
      That is, in the first 15 mins, we'll walk them through the tutorial as written. We will gloss over the parts where the tutorial shows how to compile MrXL using the eDSL, since that's really best for working through yourself. Indeed, they'll spend 45 mins adding `map` to the DSL.

3. Speaking of later on, I propose that we change

    > 11.25am-12.25pm | Hardware Performance 101 / MrXL competition
    
    into three tracks.
    
    1. Reduction trees, for those interested in speedup. 
    2. Arbitration and multi-banking, for those interested in that.
    3. Office hours, for those just wanting to backfill about what the heck is going on.

anshumanmohan, 2023-06-05:
@rachitnigam if this is okay, I can make the changes. I can also move `fud` up, if that's of interest.

priyasrikumar, 2023-06-05:
I think moving `fud` up might be good since folks will have just finished the language tutorial? Maybe we swap Pollen and `fud`?

anshumanmohan, 2023-06-05:
I propose this

Time | Topic (Presenter: verbose)
-- | --
9am-9.05am | Introduction to Calyx (Rachit: fluff)
9.05am-9.20am | Setting Up (Rachit: meet your (virtual) machine, run the runt/fud checks, group people so they have machines even if shared, etc.)
9.20am-9.55am | Your first Calyx Program (Rachit: from the getting started guide)
9:55am-10am | `fud` (Priya: playing up how scary the orchestration actually is)
10am-10:05am | Cider (Griffin: from A exam)
10:05am-10:10am | Pollen: A Pangenome Analysis DSL (Susan: stressing the don't-need-to-know-architecture point)
10:10am-10.20am | MrXL: A map-reduce frontend (Anshuman: along with a lesson about banking) 
10.20am-11am | MrXL: implementing `map`. (on their own)
11am-11:20am | Break
11.20am-11:30 | Quick lesson about reduction trees and arbitration (Anshuman: slides, handouts)
11.30pm - 12:20pm | MrXL competition: reduction trees, arbitration, or office hours (on their own) 
12.20pm-12.30pm | Closing remarks & Competition Winners (Rachit)

priyasrikumar, 2023-06-05:
This is nice!! Frontloading all the not-MrXL stuff is awesome and gives folks contextual continuity for independent work time. Totally in favor :D

anshumanmohan, 2023-06-05:
One issue is that it violates one of our original principles: we now talk at them for over an hour before they get cracking. If we wanted to soothe that, I suppose we could move Cider (and even `fud`) to the end of the day, just before closing.

anshumanmohan, 2023-06-05:
@rachitnigam, now that we are leaning on the Docker a good bit more, do you think you could squeeze the "setting up" stuff down by a few minutes and yield that time to `map` implementation?

rachitnigam, 2023-06-06:
So, the reason "setting up" is that long is because it might take people a while to download the docker image and we would need to help people who run into problem. It's already quite short so I'd feel a little iffy about making it shorter.

anshumanmohan, 2023-06-06:
Ah yes, both very fair! I've adjusted the proposed schedule [above](https://github.com/cucapra/calyx/issues/1546#issuecomment-1577589784) with these changes, see what you think?

rachitnigam, 2023-06-06:
LGTM! Merge the changes on the website!

closed 2023-06-07 by rachitnigam