calyxir/calyx #1559: Tutorial: tight, successful `fud check` invocation

anshumanmohan, 2023-06-08:
At the last meeting we spoke about having folks run a tighter incantation of `fud check` that passes 100%, with no warnings. Just making an issue so we don't forget. If you pull the Docker right now and run `fud check`, here's what's missing:

```
stages.synth-verilog.exec:
 ✖ vivado not installed.

stages.vivado-hls.exec:
 ✖ vivado_hls not installed.

stages.mrxl-data.exec:
 ✖ mrxl-data not installed.

synth-verilog, vivado-hls, mrxl-data were not installed correctly.
```

Interestingly, running
```
fud register mrxl -p frontends/mrxl/fud/mrxl.py
```
(as suggested in the [MrXL doc](http://localhost:3000/tutorial/frontend-tut.html#installing-mrxl)) has the right output:
```
Registering external script: mrxl
  - Registering stage `mrxl'.
  - Registering stage `mrxl-data'.
 ```
but `fud check` still reports the stage missing.

rachitnigam, 2023-06-08:
This is because `mrxl-data` is a fake binary that isn't supposed to exist (@susan-garry clarified this). We should change the `defaults` in the `mrxl-data` stage to not have the `exec` field. We will also need to rebuild the container after this change.

rachitnigam, 2023-06-08:
See #1560

anshumanmohan, 2023-06-08:
Nice, many thanks!

So now re: the tight `fud check` invocation. My understanding is that you want this (and, relatedly, you want `runt -i 'core | mrxl'` to be run by the participants inside of their Docker containers as a sanity check. Is that about right? I'm not opposed, but I sorta don't see what it will buy us. I say more [here](https://github.com/cucapra/calyx/pull/1556#issuecomment-1583197944).

sampsyo, 2023-06-09:
FWIW, I'm not sure it's possible (happy to be proven wrong) with current fud to suppress the messages about Vivado in the `fud check` output. The path of least resistance may be documentation: i.e., to tell people "you should see green checks everywhere except for next to `vivado` and `vivado_hls`. We won't be using those."

rachitnigam, 2023-06-09:
It is if you specify the stages you want checked: https://github.com/cucapra/filament-evaluation/blob/07ecae4b63b0fdc893c8444ccbe582fd0590c02c/scripts/configure-fud.sh#L53

sampsyo, 2023-06-10:
That's awesome!

PR #1565 (2023-06-10): Add further sanity checks to docker installation instructions
When running the tutorial inside our Docker, I am now having them run
- `runt -i 'core|mrxl'` 
- `fud check verilog jq mrxl interpreter`

If anyone sees a reason to check additional `fud` stages, speak now!

I have check that these do work cleanly inside the latest Docker image. I have not tested these in any other form of Calyx installation (crate or built-from-source).

Closes #1559.
closed 2023-06-10 by sampsyo