calyxir/calyx #1584: Consider adding a non-static repeat loop

andrewb1999, 2023-07-04:
Currently Calyx supports `static repeat n {...}` ops which run the loop body a set number of times. This syntax has allowed static compilation to produce more efficient static FSMs. Even though this syntax was originally designed to help with static compilation, I wonder if it would be useful to have a non-static (dynamic) repeat loop `repeat n {...}` just for syntax reasons. This new repeat loop is just syntax sugar for a dynamic while loop that runs a specified number of times, but will make lowering to Calyx cleaner in some circumstances. It will also bring static and dynamic syntax to parity and potentially help with static promotion.

rachitnigam, 2023-07-05:
This would also make promotion of dynamic code to static easier since we'd be able to promote loop regions as well.

@calebmkim mentioned that the analysis code might have to account for another kind of loop but to a first-order approximation, we can reuse the same reasoning we have for `while` loops.

rachitnigam, 2023-07-08:
@sampsyo @EclecticGriffin this seems reasonable to me as an addition to the language. The benefit to `static-promotion` is reason enough for me but wanted to get your thoughts on this before finalizing the proposal.

@andrewb1999 this is likely to be accepted; do you want to try to add an implementation for it in a PR?

sampsyo, 2023-07-08:
Yes! It seems like a really nice "invariant" to preserve that all static control statements have a dynamic equivalent. That way, a frontend can always generate static code by generating a fully dynamic program and relying on auto-static-ification.

calebmkim, 2023-07-08:
I wouldn't mind trying to implement this, since I already have in mind how it might affect the analyses. But @andrewb1999 feel free to take it if this is something you'd want to do.

andrewb1999, 2023-07-08:
> I wouldn't mind trying to implement this, since I already have in mind how it might affect the analyses. But @andrewb1999 feel free to take it if this is something you'd want to do.

Feel free to work on it if you have time. I likely won't have time for a few more days anyways.

calebmkim, 2023-07-13:
I think I'll take this, since I've already started it a bit. I think it could help `static-promotion` when evaluating the SCF->Calyx polybench benchmarks.

calebmkim, 2023-07-17:
Closed by https://github.com/cucapra/calyx/pull/1600

closed 2023-07-21 by andrewb1999