calyxir/calyx #1587: Documentation: need documetation in how to use how to generate Verilog from CIRCT generated Calyx

ManelMCCS, 2023-07-07:


rachitnigam, 2023-07-07:
It's here: https://docs.calyxir.org/fud/circt.html
Once you have a native Calyx program, you can [install the `calyx` compiler`](https://docs.calyxir.org/intro.html#installing-the-crate-to-use-but-not-extend-calyx) and use that to generate Verilog code.

Feel free to reopen the issue if you need more help with it

closed 2023-07-07 by rachitnigam

rachitnigam, 2023-07-10:
Ah, can you change the —from futil to —from calyx instead?

ManelMCCS, 2023-07-10:
```
root@ab83ec2abc5a:/home/calyx# fud exec examples/futil/simple.expect --from calyx --to verilog
[fud] ERROR: `/home/calyx/target/debug/calyx -l /home/calyx -b verilog' failed:
=====STDERR=====
thread 'main' panicked at 'static component main does not have static control. This should have been checked in ast_to_ir', calyx-opt/src/passes/well_formed.rs:304:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

=====STDOUT=====

I'm still getting errors
```

rachitnigam, 2023-07-10:
Looking into it now

rachitnigam, 2023-07-10:
Ah, it's because you're running it on a test output file. The correct command to run:
```
fud exec examples/futil/simple.futil --from calyx --to verilog
```

Where did you get that command from? If it's in the docs, we can fix it.

ManelMCCS, 2023-07-10:
Thanks, its workings, I will continuing to try to generate Verilog from upstream MLIR using Calyx and CIRCT

rachitnigam, 2023-07-10:
Awesome! Feel free to open more issues if you run into any problems. We’re happy to help

ManelMCCS, 2023-07-28:
I did it, if you add this to this line to the documentation of using CIRCT as a Calyx front-end is could be very useful for beginners!
Feel free to close the issue!
Thanks!

rachitnigam, 2023-07-31:
Thanks! It's already been fixed!